react-native-camera/react-native-camera #1230: iOS: Warning Setting focusDepth isn't supported

n1ru4l, 2018-02-12:
### Which implementation are you using

`RNCamera`

### Steps to reproduce
1. Upgrade to RNCamera
2. Create a view with the following props: ` ref={cam => {
          this.camera = cam
        }}`
        `type={this.state.cameraType}`
3. Start your app and navigate to the camera view

### Expected behaviour
The warning should not appear since I am not setting any property that affects the focusDepth.

### Actual behaviour

Yellow warning box with the message `-[RNCamera updateFocusDepth]: Setting focusDepth isn't supported for this camera device` appears

### Environment
- **Node.js version**: `v8.9.4`
- **React Native version**: `0.52.2`
- **React Native platform + platform version**: iOS 10.0

### react-native-camera
**Version**: `1.0.0`

mbodge, 2018-02-26:
I'm still getting this even with the merged code. I'm using the front camera, so maybe that has something to do with it?

albertadevor, 2018-03-16:
Also experiencing this issue - it occurs when I switch from back camera to front camera on iOS. Also using v0.52.2 of RN.

jgfidelis, 2018-04-15:
Hmmm... probably need to check in the if condition, if it is a front camera. I will try tomorrow.

AlexBlagodaryov, 2018-04-18:
Hello, any updates for this issue?

sibelius, 2018-04-18:
@AlexBlagodaryov feel free to send a PR fixing it

delta98, 2018-04-19:
@sibelius Does this look like it will solve the problem? 

`(self.autoFocus < 0 || device.focusMode != RNCameraAutoFocusOff || device.position == RNCameraTypeFront)`

PR #1243 (2018-02-14): fix(focusWarning): fix focus depth warning being shown for ios.
Closes #1230