react-native-camera/react-native-camera #1231: Build failed for RNCameraUtils

chmielot, 2018-02-12:
### Which implementation are you using

*RNCamera*

### Steps to reproduce
1. Follow the migration guide to RNCamera
2. I don't need FaceDetection, so deleted the folder in node_modules and from XCode project
3. `react-native run-ios`

### Expected behaviour
Build succeeds

### Actual behaviour
Build fails:

```
CompileC /Users/me/Code/ios/build/Build/Intermediates/RNCamera.build/Debug-iphonesimulator/RNCamera.build/Objects-normal/x86_64/RNCameraUtils.o RN/RNCameraUtils.m normal x86_64 objective-c com.apple.compilers.llvm.clang.1_0.compiler
    cd /Users/me/Code/node_modules/react-native-camera/ios
    export LANG=en_US.US-ASCII
    export PATH="/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/usr/bin:/Applications/Xcode.app/Contents/Developer/usr/bin:/usr/local/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/MacGPG2/bin:/Users/cht/Library/Android/sdk/tools"
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -x objective-c -arch x86_64 -fmessage-length=0 -fdiagnostics-show-note-include-stack -fmacro-backtrace-limit=0 -std=gnu99 -fobjc-arc -fmodules -fmodules-cache-path=/Users/me/Code/ios/build/ModuleCache -fmodules-prune-interval=86400 -fmodules-prune-after=345600 -fbuild-session-file=/Users/me/Code/ios/build/ModuleCache/Session.modulevalidation -fmodules-validate-once-per-build-session -Wnon-modular-include-in-framework-module -Werror=non-modular-include-in-framework-module -Wno-trigraphs -fpascal-strings -O0 -Wno-missing-field-initializers -Wno-missing-prototypes -Werror=return-type -Wunreachable-code -Wno-implicit-atomic-properties -Werror=deprecated-objc-isa-usage -Werror=objc-root-class -Wno-arc-repeated-use-of-weak -Wduplicate-method-match -Wno-missing-braces -Wparentheses -Wswitch -Wunused-function -Wno-unused-label -Wno-unused-parameter -Wunused-variable -Wunused-value -Wempty-body -Wconditional-uninitialized -Wno-unknown-pragmas -Wno-shadow -Wno-four-char-constants -Wno-conversion -Wconstant-conversion -Wint-conversion -Wbool-conversion -Wenum-conversion -Wshorten-64-to-32 -Wpointer-sign -Wno-newline-eof -Wno-selector -Wno-strict-selector-match -Wundeclared-selector -Wno-deprecated-implementations -DDEBUG=1 -DOBJC_OLD_DISPATCH_PROTOTYPES=0 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator10.2.sdk -fasm-blocks -fstrict-aliasing -Wprotocol -Wdeprecated-declarations -mios-simulator-version-min=8.2 -g -Wno-sign-conversion -Wno-infinite-recursion -fobjc-abi-version=2 -fobjc-legacy-dispatch -iquote /Users/me/Code/ios/build/Build/Intermediates/RNCamera.build/Debug-iphonesimulator/RNCamera.build/RNCamera-generated-files.hmap -I/Users/me/Code/ios/build/Build/Intermediates/RNCamera.build/Debug-iphonesimulator/RNCamera.build/RNCamera-own-target-headers.hmap -I/Users/me/Code/ios/build/Build/Intermediates/RNCamera.build/Debug-iphonesimulator/RNCamera.build/RNCamera-all-target-headers.hmap -iquote /Users/me/Code/ios/build/Build/Intermediates/RNCamera.build/Debug-iphonesimulator/RNCamera.build/RNCamera-project-headers.hmap -I/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/include -I../../../ios -I../../../ios/build -I../../../ios/template -I../../../ios/template-tvOS -I../../../ios/template-tvOSTests -I../../../ios/templateTests -I../../../ios/build/Build -I../../../ios/build/Logs -I../../../ios/build/Logs/Build -I/Users/me/Code/ios/build/Build/Intermediates/RNCamera.build/Debug-iphonesimulator/RNCamera.build/DerivedSources/x86_64 -I/Users/me/Code/ios/build/Build/Intermediates/RNCamera.build/Debug-iphonesimulator/RNCamera.build/DerivedSources -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/include -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/template.app -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/template.app.dSYM -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/templateTests.xctest.dSYM -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/include/cxxreact -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/include/double-conversion -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/include/fishhook -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/include/jschelpers -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/include/jsinspector -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/include/privatedata -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/include/RCTAnimation -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/include/RCTBlob -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/include/RCTImage -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/include/RCTText -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/include/React -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/include/yoga -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/template.app/_CodeSignature -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/template.app/Frameworks -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/template.app/PlugIns -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/template.app.dSYM/Contents -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/templateTests.xctest.dSYM/Contents -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/template.app/PlugIns/templateTests.xctest -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/template.app.dSYM/Contents/Resources -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/templateTests.xctest.dSYM/Contents/Resources -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/template.app/PlugIns/templateTests.xctest/_CodeSignature -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/template.app.dSYM/Contents/Resources/DWARF -F/Users/me/Code/ios/build/Build/Products/Debug-iphonesimulator/templateTests.xctest.dSYM/Contents/Resources/DWARF -F../../../ios -F../../../ios/build -F../../../ios/template -F../../../ios/template-tvOS -F../../../ios/template-tvOSTests -F../../../ios/templateTests -F../../../ios/build/Build -F../../../ios/build/Logs -F../../../ios/build/Logs/Build -MMD -MT dependencies -MF /Users/me/Code/ios/build/Build/Intermediates/RNCamera.build/Debug-iphonesimulator/RNCamera.build/Objects-normal/x86_64/RNCameraUtils.d --serialize-diagnostics /Users/me/Code/ios/build/Build/Intermediates/RNCamera.build/Debug-iphonesimulator/RNCamera.build/Objects-normal/x86_64/RNCameraUtils.dia -c /Users/me/Code/node_modules/react-native-camera/ios/RN/RNCameraUtils.m -o /Users/me/Code/ios/build/Build/Intermediates/RNCamera.build/Debug-iphonesimulator/RNCamera.build/Objects-normal/x86_64/RNCameraUtils.o
/Users/me/Code/node_modules/react-native-camera/ios/RN/RNCameraUtils.m:14:43: error: expected a type
+ (AVCaptureDevice *)deviceWithMediaType:(AVMediaType)mediaType preferringPosition:(AVCaptureDevicePosition)position
                                          ^
1 error generated.

** BUILD FAILED **

The following build commands failed:
  CompileC /Users/me/Code/ios/build/Build/Intermediates/RNCamera.build/Debug-iphonesimulator/RNCamera.build/Objects-normal/x86_64/RNCameraUtils.o RN/RNCameraUtils.m normal x86_64 objective-c com.apple.compilers.llvm.clang.1_0.compiler
(1 failure)
```

### Environment
- **Node.js version**: 9.5.0
- **React Native version**: 0.53.0
- **React Native platform + platform version**: Simulator iOS 9.3, iPhone 6

### react-native-camera
**Version**: master (today)

jgfidelis, 2018-02-12:
What folder did you delete?

paynd, 2018-02-13:
Got the same issue.
@chmielot Just deleted folder with command
`rm -rf node_modules/react-native-camera/ios/FaceDetector`
and
`cp node_modules/react-native-camera/postinstall_project/projectWithoutFaceDetection.pbxproj node_modules/react-native-camera/ios/RNCamera.xcodeproj/project.pbxproj`

I documentation we use foldermname `RCTCamera.xcodeproj` there are no such folder, I changed it to RNCamera.xcodeproj.

chmielot, 2018-02-13:
@jgfidelis I did: `rm -rf node_modules/react-native-camera/ios/FaceDetector`
and removed `FaceDetection` from the xcodeproj by hand.

@paynd Glad that there is someone else having the same issue. I spent about 3 hours experimenting and reinstalling with no success.

chmielot, 2018-02-13:
@paynd , no I didn't because the README says:

>If you want to make this automatic, you can add a postinstall script to your app package.json. Inside the postinstall_project there is a xcode project ready with the folder removed (we opened xcode, removed the folder from the project and copied the resulting project file).

I don't need to have this automatic and what the script does is apparently the same I did by hand.

I am still new to react-native. Do you mean that by applying the postinstall script the issue can be solved? .. and that maybe I did a rebuilt and the FaceDetection was restored in the xcode proj (I can have a look in about 10hours from now)?
I am also not sure if I understood: Have you been able to solve the issue or are you still experiencing it?

paynd, 2018-02-13:
@chmielot no, I still experiencing it too.

jgfidelis, 2018-02-13:
Yes, if you `npm install` you will have to delete it once again, that is why we recommend making a postintall script, to automate the process of deleting it, so you will not have to worry about it.

If you delete the folder with `rm -rf` and copy the xcode project correctly over the old one, it should build.

We are investigating ways to make the installation process better.

densa, 2018-02-13:
@chmielot that's how I fix it for xcode 8
Open your ios project. And add ImageIO.framework into linked frameworks for you target. Made small changes in react-native-camera ios code (see [here](https://github.com/react-native-community/react-native-camera/compare/master...densa:master))
If you need can for and update or use my fork in package.json like this `"react-native-camera": "https://github.com/densa/react-native-camera.git#9a03a2c"`

jgfidelis, 2018-02-13:
It might be some recursive imports.

Could you try a branch where we removed some bad imports?

`"react-native-camera": "git+https://github.com/jgfidelis/react-native-camera#a5b9f7e717bc11aad9a8e5d9e9a449ad7fd9c9fa"`

Please clean your xcode project (cmd+shift+k) before trying to build.

chmielot, 2018-02-14:
@jgfidelis I reverted everything back to a working version with react-native-camera 0.12.0.
From there I updated to the branch you gave me by setting this in `package.json` and calling `npm install react-native-camera`. After that I followed the migration guide again, made sure that `FaceDetection` does not exist in the Libraries, cleaned the build and ran `react-native run-ios`. Unfortunately I still got the error:

```
CompileC /Users/cht/Code/template/ios/build/Build/Intermediates/RNCamera.build/Debug-iphonesimulator/RNCamera.build/Objects-normal/x86_64/RNCameraManager.o RN/RNCameraManager.m normal x86_64 objective-c com.apple.compilers.llvm.clang.1_0.compiler
    cd /Users/cht/Code/template/node_modules/react-native-camera/ios
    export LANG=en_US.US-ASCII
    export PATH="/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/usr/bin:/Applications/Xcode.app/Contents/Developer/usr/bin:/usr/local/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/MacGPG2/bin:/Users/cht/Library/Android/sdk/tools:/Users/cht/Library/Android/sdk/platform-tools"
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -x objective-c -arch x86_64 -fmessage-length=0 -fdiagnostics-show-note-include-stack -fmacro-backtrace-limit=0 -std=gnu99 -fobjc-arc -fmodules -fmodules-cache-path=/Users/cht/Code/template/ios/build/ModuleCache -fmodules-prune-interval=86400 -fmodules-prune-after=345600 -fbuild-session-file=/Users/cht/Code/template/ios/build/ModuleCache/Session.modulevalidation -fmodules-validate-once-per-build-session -Wnon-modular-include-in-framework-module -Werror=non-modular-include-in-framework-module -Wno-trigraphs -fpascal-strings -O0 -Wno-missing-field-initializers -Wno-missing-prototypes -Werror=return-type -Wunreachable-code -Wno-implicit-atomic-properties -Werror=deprecated-objc-isa-usage -Werror=objc-root-class -Wno-arc-repeated-use-of-weak -Wduplicate-method-match -Wno-missing-braces -Wparentheses -Wswitch -Wunused-function -Wno-unused-label -Wno-unused-parameter -Wunused-variable -Wunused-value -Wempty-body -Wconditional-uninitialized -Wno-unknown-pragmas -Wno-shadow -Wno-four-char-constants -Wno-conversion -Wconstant-conversion -Wint-conversion -Wbool-conversion -Wenum-conversion -Wshorten-64-to-32 -Wpointer-sign -Wno-newline-eof -Wno-selector -Wno-strict-selector-match -Wundeclared-selector -Wno-deprecated-implementations -DDEBUG=1 -DOBJC_OLD_DISPATCH_PROTOTYPES=0 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator10.2.sdk -fasm-blocks -fstrict-aliasing -Wprotocol -Wdeprecated-declarations -mios-simulator-version-min=8.2 -g -Wno-sign-conversion -Wno-infinite-recursion -fobjc-abi-version=2 -fobjc-legacy-dispatch -iquote /Users/cht/Code/template/ios/build/Build/Intermediates/RNCamera.build/Debug-iphonesimulator/RNCamera.build/RNCamera-generated-files.hmap -I/Users/cht/Code/template/ios/build/Build/Intermediates/RNCamera.build/Debug-iphonesimulator/RNCamera.build/RNCamera-own-target-headers.hmap -I/Users/cht/Code/template/ios/build/Build/Intermediates/RNCamera.build/Debug-iphonesimulator/RNCamera.build/RNCamera-all-target-headers.hmap -iquote /Users/cht/Code/template/ios/build/Build/Intermediates/RNCamera.build/Debug-iphonesimulator/RNCamera.build/RNCamera-project-headers.hmap -I/Users/cht/Code/template/ios/build/Build/Products/Debug-iphonesimulator/include -I../../../ios -I../../../ios/build -I../../../ios/template -I../../../ios/template-tvOS -I../../../ios/template-tvOSTests -I../../../ios/templateTests -I../../../ios/build/Build -I../../../ios/build/Logs -I../../../ios/build/Logs/Build -I/Users/cht/Code/template/ios/build/Build/Intermediates/RNCamera.build/Debug-iphonesimulator/RNCamera.build/DerivedSources/x86_64 -I/Users/cht/Code/template/ios/build/Build/Intermediates/RNCamera.build/Debug-iphonesimulator/RNCamera.build/DerivedSources -F/Users/cht/Code/template/ios/build/Build/Products/Debug-iphonesimulator -F/Users/cht/Code/template/ios/build/Build/Products/Debug-iphonesimulator -F/Users/cht/Code/template/ios/build/Build/Products/Debug-iphonesimulator/include -F/Users/cht/Code/template/ios/build/Build/Products/Debug-iphonesimulator/template.app -F/Users/cht/Code/template/ios/build/Build/Products/Debug-iphonesimulator/template.app.dSYM -F/Users/cht/Code/template/ios/build/Build/Products/Debug-iphonesimulator/templateTests.xctest.dSYM -F/Users/cht/Code/template/ios/build/Build/Products/Debug-iphonesimulator/include/cxxreact -F/Users/cht/Code/template/ios/build/Build/Products/Debug-iphonesimulator/include/double-conversion -F/Users/cht/Code/template/ios/build/Build/Products/Debug-iphonesimulator/include/fishhook -F/Users/cht/Code/template/ios/build/Build/Products/Debug-iphonesimulator/include/jschelpers -F/Users/cht/Code/template/ios/build/Build/Products/Debug-iphonesimulator/include/jsinspector -F/Users/cht/Code/template/ios/build/Build/Products/Debug-iphonesimulator/include/privatedata -F/Users/cht/Code/template/ios/build/Build/Products/Debug-iphonesimulator/include/RCTAnimation -F/Users/cht/Code/template/ios/build/Build/Products/Debug-iphonesimulator/include/RCTBlob -F/Users/cht/Code/template/ios/build/Build/Products/Debug-iphonesimulator/include/RCTImage -F/Users/cht/Code/template/ios/build/Build/Products/Debug-iphonesimulator/include/RCTText -F/Users/cht/Code/template/ios/build/Build/Products/Debug-iphonesimulator/include/React -F/Users/cht/Code/temp [... cut ...]
+ (AVCaptureDevice *)deviceWithMediaType:(AVMediaType)mediaType preferringPosition:(AVCaptureDevicePosition)position
                                          ^
1 error generated.

** BUILD FAILED **

The following build commands failed:
    CompileC /Users/cht/Code/template/ios/build/Build/Intermediates/RNCamera.build/Debug-iphonesimulator/RNCamera.build/Objects-normal/x86_64/RNCameraUtils.o RN/RNCameraUtils.m normal x86_64 objective-c com.apple.compilers.llvm.clang.1_0.compiler
(1 failure)
```

jgfidelis, 2018-02-14:
What XCode are you using?

chmielot, 2018-02-14:
Version 8.2.1 (8C1002)

sibelius, 2018-02-14:
please use Xcode 9.2

chmielot, 2018-02-14:
I can't, because Xcode 9.2 doesn't run on OSX 10.11.6 (El Capitan) which is the last possible OSX version for my Mac. :(

jgfidelis, 2018-02-14:
Please run this command:
```
rm -rf node_modules && rm package-lock.json && npm install
```

Run the post install script:
```
#!/bin/bash
rm -rf node_modules/react-native-camera/ios/FaceDetector
cp node_modules/react-native-camera/postinstall_project/projectWithoutFaceDetection.pbxproj node_modules/react-native-camera/ios/RNCamera.xcodeproj/project.pbxproj
```

Try to compile. If it does not work, try to open xcode, press `cmd+shift+k` and then build with `cmd+b`.

densa, 2018-02-14:
@chmielot did you see my comment about xcode 8 ?

chmielot, 2018-02-14:
@jgfidelis I followed your instructions, but unfortunately got the same result, this time a screenshot from XCode:
[image]

I also wanted to mention, that before I upgraded `react-native-camera` from 0.12.0 to the current version, I made sure that I'm able to build for my Android device. Everything worked well. After the upgrade I get the following during build:

```
~/Code/template/android$ ./gradlew clean
Incremental java compilation is an incubating feature.
File /Users/cht/.android/repositories.cfg could not be loaded.

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':app'.
> Could not resolve all dependencies for configuration ':app:_debugApkCopy'.
   > Could not find com.android.support:exifinterface:26.0.2.
     Searched in the following locations:
         file:/Users/cht/Library/Android/sdk/extras/android/m2repository/com/android/support/exifinterface/26.0.2/exifinterface-26.0.2.pom
         file:/Users/cht/Library/Android/sdk/extras/android/m2repository/com/android/support/exifinterface/26.0.2/exifinterface-26.0.2.jar
         file:/Users/cht/Code/sellmycar/android/sdk-manager/com/android/support/exifinterface/26.0.2/exifinterface-26.0.2.jar
     Required by:
         template:app:unspecified > template:react-native-camera:unspecified
   > Could not find com.android.support:support-v4:26.0.1.
     Searched in the following locations:
         file:/Users/cht/Library/Android/sdk/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.pom
         file:/Users/cht/Library/Android/sdk/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
         file:/Users/cht/Code/sellmycar/android/sdk-manager/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
     Required by:
         template:app:unspecified > com.android.support:appcompat-v7:23.0.1
   > Could not find com.android.support:support-annotations:26.0.1.
     Searched in the following locations:
         file:/Users/cht/Library/Android/sdk/extras/android/m2repository/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.pom
         file:/Users/cht/Library/Android/sdk/extras/android/m2repository/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.jar
         file:/Users/cht/Code/sellmycar/android/sdk-manager/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.jar
     Required by:
         template:app:unspecified > template:react-native-camera:unspecified > com.github.react-native-community:cameraview:d5d9b0d925494ef451ce3eef3fdf14cc874d9baa
   > Could not find com.android.support:support-v4:26.0.1.
     Searched in the following locations:
         file:/Users/cht/Library/Android/sdk/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.pom
         file:/Users/cht/Library/Android/sdk/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
         file:/Users/cht/Code/sellmycar/android/sdk-manager/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
     Required by:
         template:app:unspecified > template:react-native-camera:unspecified > com.github.react-native-community:cameraview:d5d9b0d925494ef451ce3eef3fdf14cc874d9baa
   > Could not find com.android.support:support-v4:26.0.1.
     Searched in the following locations:
         file:/Users/cht/Library/Android/sdk/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.pom
         file:/Users/cht/Library/Android/sdk/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
         file:/Users/cht/Code/sellmycar/android/sdk-manager/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
     Required by:
         template:app:unspecified > com.google.android.gms:play-services-vision:10.2.0 > com.google.android.gms:play-services-basement:10.2.0

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 14.633 secs
```

So the Android build isn't working as well.

@densa I was able to build from your branch with the instructions you provided. I see warnings in the Simulator "[RNCamera updateFocusDepth]: Setting focusDepth isn't supported for this camera device" (iPhone6, iOS 9.3) and the images are in square format (200x200) as opposed to RCTCamera's images, that were always in portrait mode. None of these two issues are related to your changes?

jgfidelis, 2018-02-14:
@densa hey, your fix looks good, is it something xcode 8 related? Do you know what is happening?

densa, 2018-02-14:
@jgfidelis yes, it's related to xcode 8. 
1 - AVMediaType available from ios 11 (xcode 8 - ios 10)
2 - I don't know why but xcode 8 won't compile RNImageUtils w/o import ImageIO.h
3 - xcode 8 won't recognize AVMetadataObjectType so I just change it to nsstring

sibelius, 2018-02-14:
we should only support latest versions of xcode and react-native, 

it would make easy for us to maintain this package

we could have a branch xcode8 in this repo if you want to send a pr for this @densa

closed 2018-02-14 by sibelius

chmielot, 2018-02-15:
Supporting only the latest version of react-native - fine.
Supporting only the latest version of XCode - sad. - Need to buy new hardware then :(

Any comments on the failed build for Android? Is this related, should I open a new issue? Please advice.

sibelius, 2018-02-15:
For android follow README, u r missing jitpack and google() on build.gradle

chmielot, 2018-02-15:
I followed the README:

```
// android/app/build.gradle
dependencies {
    compile (project(':react-native-camera')) {
      exclude group: "com.google.android.gms"
    }
    compile ("com.google.android.gms:play-services-vision:10.2.0") {
      force = true;
    }
    compile project(':react-native-vector-icons')
    compile project(':react-native-i18n')
...
```

and

```
// android/build.gradle
allprojects {
    repositories {
        mavenLocal()
        jcenter()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
        maven { url "https://jitpack.io" }
    }
}
```

I repeated all of the steps again. Can you give advice, please?

sibelius, 2018-02-15:
// android/build.gradle
allprojects {
    repositories {
        mavenLocal()
        jcenter()
       google()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
        maven { url "https://jitpack.io" }
    }
}

use the latest gradle version as well

try to run using Android Studio, Android Studio can fix this issues easily

chmielot, 2018-02-17:
Thank you for your patience @sibelius . I made it work using Android Studio. Android Studio updated gradle and a few other things and I was able to build it afterwards. 
The only thing I noticed compared to RCTCamera is that it takes longer after pressing the shutter button for the bound action (nothing special, just update a property in redux store and navigate back) to execute (even 2-3 seconds).

jqn, 2018-03-07:
I had the same problem and the android studio solution worked also.

qumberali22, 2018-03-14:
@jgfidelis thanks. It worked for me.