react-native-camera/react-native-camera #1233: Android: RNCamera preview stops after async task

changx03, 2018-02-13:
I switched from Camera to RNCamera. However, the startPreview() does no longer exist in RNCamera. 
After some async tasks in onBarCodeRead function, the preview stops working, and there is no way to restart the camera.
What's going on here? 

```
<RNCamera
          autoFocus={RNCamera.Constants.AutoFocus.on}
          flashMode={RNCamera.Constants.FlashMode.off}
          onBarCodeRead={this._onBarCodeRead}
          playSoundOnCapture={false}
          style={styles.cameraView}
          type={RNCamera.Constants.Type.back}
          ref={ref => { this._camera = ref; }}
        >
  // UI goes here...
</RNCamera>
```

```
private _onBarCodeRead = e => {
      // this._camera.stopPreview();
      Sound.play(success => {
        if (success) {
          // check bin scan list
          SubmissionStore.addBinScan(e.data).then(() => {
            this._previousQRCode = e.data;
            // this._camera.startPreview();
          });
        }
      });
    }
};
```

K-Leon, 2018-02-17:
I think this is related to #1259

sibelius, 2018-02-17:
the new RNCamera does not have this methods

we should implement

can something take a look in this? 

just follow this https://github.com/react-native-community/react-native-camera/pulls?q=is%3Apr+stopPreview+is%3Aclosed

closed 2019-06-08 by stale[bot]