react-native-camera/react-native-camera #1234: RNCamera not working on android emulator (black screen)

cvvspb, 2018-02-13:
### Which implementation are you using

*RNCamera*

### Steps to reproduce
`<RNCamera`
`ref={(cam) => { this.camera = cam; }}`
`type={RNCamera.Constants.Type.back}`
`// onBarCodeRead={this.barcodeRead}`
`barCodeTypes={[RNCamera.Constants.BarCodeType.qr]}`
`style={styles.preview}`
`>`
`<Text style={styles.capture} onPress={this.emulateScan}>[id: 4 code: 5K0C3H]</Text>`
`<TouchableOpacity onPress={() => Alert.alert('Alert', 'Keyboard pressed')}>`
`	<Image source={imgKeyboard} style={[styles.icon]} />`
`</TouchableOpacity>`
`</RNCamera>`

### Expected behaviour
Should display a preview screen

### Actual behaviour
This code does work on a device in debug mode as expected, but on an emulator displays a black screen.
Tested on Android (4.1, 6.0, 7.1.1) ABI x86 with Google API

### Environment
- **Node.js version**: 8.9.0
- **React Native version**: 0.50.0
- **React Native platform + platform version**: Android (4.1, 6.0, 7.1.1) ABI x86 with Google API

### react-native-camera
**Version**: "^1.0.0"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

jgfidelis, 2018-02-13:
We have to take a look into it but emulators were not a priority to 1.0.0

rasom, 2018-02-14:
This happens on real devices as well, at least on Android 5.0.1, 6.0.1

rasom, 2018-02-14:
It actually works on real device in debug mode, but can be reproduced in release mode.

sibelius, 2018-02-14:
This could be a bug with android 6 permissions

asemiankevich, 2018-02-14:
I can see this on Android 7

tobob, 2018-02-21:
on all my emulators I have same issue, however I did check it on 3 real devices and it works ok, however when I install expo camera app using expo on one of emulators it making images correctly

tobob, 2018-02-21:
```
new ResolveTakenPictureAsyncTask(byteBuffer.array(), promise, options).execute();
```

this line fails, it on only called on Emulators

Tsiniloiv, 2018-03-05:
I'm also getting this error on my emulator, and also on one of my test devices, though we've so far been unable to reproduce the error intentionally on a physical device.

tutts, 2018-03-19:
We're finding this issue on Android 7 and 8 on a release build too. Most notably when we use React Navigation and already have a camera (scanning enabled) on one view and then attempt to load another camera on top later on in the navigation stack, wondering if there's a reference issue somewhere as both will be mounted until popped off the stack...

rasom, 2018-03-23:
as @tutts mentioned, it can be reproduced in release build in real device

CPCherrington, 2018-04-04:
Im getting this issue also

waitopiggu, 2018-04-15:
I've tracked it down to [here](https://github.com/react-native-community/react-native-camera/blob/a64d31eb4591c60f89d1ce444f1944232c057faa/android/src/main/java/org/reactnative/camera/RNCameraView.java#L439).

If you remove the fingerprint check, the webcam does indeed start, at least, on my machine.

```
//if (!Build.FINGERPRINT.contains("generic")) {
  start();
//}
```

I'm sure this check was put there for a reason. It's been there since the first history entry `RNCameraView.java`.

avshenuk, 2018-04-15:
> @sibelius <https://github.com/sibelius> Do you know who originally wrote
> it? It would be good to find out why they put it in before removing it.

As I know this check actually means “if this is emulator” because their
fingerprints contain “generic” in the fingerprint... so this is definitely
for a reason there :)

waitopiggu, 2018-04-15:
@avshenuk Well, I set my emulator webcam to "none" and the app didn't crash, but there was a logcat error.

`W/CameraBase: An error occurred while connecting to camera -1: Status(-8): '4: validateConnectLocked:899: No camera device with ID "-1" available'`

My remote debugger isn't working for emulated devices, so I can't check if React exceptions are being handled well.

sibelius, 2018-04-15:
@aalices do u know why there is this check?

aalices, 2018-04-16:
Yeah I put it by design to not render any preview when using an emulator cause iirc this was often crashing

adipascu, 2018-04-16:
It whould be nice to show something like `<Text>The camera is currently unavailable on emulators</Text>`instead of a blank screen to avoid confusion.

waitopiggu, 2018-04-17:
@aalices I'll create some AVDs with different Android versions and architectures when I get a chance. I've only been on testing on "P" x86 so far.

ChristineTzeng, 2018-05-01:
I got the same black screen on Android 7(API 25/x86) emulator.

closed 2018-05-18 by jgfidelis

mirceanis, 2018-05-18:
do you think we can get a release with this small fix in?

commit 97fad721 (2018-03-30): Merge pull request #1414 from wbroek/fix/emulator

Emulator crash fix #1234
commit d0e0d261 (2018-03-30): Emulator crash fix #1234

On the emulator it crashed because the image byte array was incorrect and no cache directory was defined so it could not be written
PR #1571 (2018-05-18): reenable camera on android emulator
This fixes #1234 
The camera is disabled on android emulator for no good reason (or the reason may have fixed itself since). This PR brings it back.

I am able to use the camera on emulators with no problem using this fix. No more black screen, Yay!