react-native-camera/react-native-camera #1236: Lag while using RNCamera to record videos.

karankapoor229, 2018-02-13:
I am facing a lag while starting and stopping the video recording using RNCamera , while with RCTCamera it is working fine.

### Steps to reproduce
**In Render Method**
`            <RNCamera
                ref={(cam) => {
                    this.camera = cam;
                }}
                type={RNCamera.Constants.Type.front}
                style={styles.preview}
                permissionDialogTitle={'Permission to use camera'}
                permissionDialogMessage={'We need your permission to use your camera phone'}
                captureAudio= {true}
                ratio = "16:9"
            >`
**On Button Click**
`takeVideo() {
        this.state.isRecording ?
            this.camera.stopRecording()
            : this.saveVideo()
        this.state.isRecording ?   this.setState({isRecording: false}) : this.setState({isRecording: true})
    }`
**Function to start and save video recording**
`saveVideo = async function() {
        if (this.camera) {
            const options = { maxDuration: 10 }
            const data = await this.camera.recordAsync(options)
            this.setState({ path: data.uri })
            this.setState({recorded: true})
        }};`

### Expected behaviour
The video recording should start instantly on button click.

### Actual behaviour
There is a significant lag after pressing the button to start recording video and same while stopping the recording.

### Environment
- **Node.js version**: 8.9.4
- **React Native version**: 0.52.0
- **React Native platform + platform version**: Android 8.0

### react-native-camera
**Version**:  "master"

jgfidelis, 2018-02-13:
Thanks for the input.

RCTCamera uses MediaRecorder to record while RNCamera uses Google's CameraView, from what I see there is nothing stopping the CameraView from starting to record on RNCamera so I think it is a problem on that class, a slow setup, I do not know. If anyone knows Android better than me and has a better opinion, it would be great.

sibelius, 2019-04-18:
RCTCamera is removed https://github.com/react-native-community/react-native-camera/issues/1890

Migrate to latest version of RNCamera

closed 2019-04-18 by sibelius

asarasua, 2019-08-08:
I'm facing the same issue, and also not using RCTCamera but RNCamera (latest version), as @karankapoor229.

baughmann, 2020-01-02:
This is still an issue. It exists on iOS but is even worse on Android. The issue remains unchanged when using a release build.

eqlion, 2021-04-01:
Still have this issue with react-native-camera 3.38.0, RN 0.62.2 and Android 10

baughmann, 2021-04-01:
@eqlion The best solution for me was to get down and dirty and make native components for the Camera. Most of your camera UI can use your normal RN components, but it's better to implement the actual camera yourself in Swift/Kotlin

eqlion, 2021-04-02:
@baughmann Could you share your solution, if it is generic? All we need is recording videos and taking pictures, and I'm proficient enough neither in Kotlin, nor in Swift :(

baughmann, 2021-04-02:
@eqlion I've been thinking about trying to re-create it at some point. Thing is, even if you're good in Swift/Objective-C and Kotlin/Java, using Native Modules and Native components is notoriously poorly documented and confusing. To my knowledge, this issue has never been addressed.

I can try to find my code and make some gists for it, but that project has been lost to time for the most part. If you want a smooth and fast camera, it is your best bet.

If a camera is not a core part of your application, I would suggest that you search for another plugin. You could also fork and modify this package to try to eliminate some of the features you do not need to make it simpler and maybe faster.