react-native-camera/react-native-camera #1237: Ambiguous resolution after pod install

ishigamii, 2018-02-13:
### Which implementation are you using

*RNCamera* 

### Steps to reproduce
1. npm install react-native-camera --save
2. react-native link react-native-camera

### Expected behaviour
Compile

### Actual behaviour
Crash on launch (ios at least) 

`This warning is caused by a @providesModule declaration with the same name across two different files.
error: bundling failed: ambiguous resolution: module `/Users/xxxx/Documents/ReactJS/xxxx/index.js` tries to require `react-native`, but there are several files providing this module. You can delete or fix them: 

  * `/Users/xxxx/Documents/ReactJS/xxxx/ios/Pods/React/package.json`
  * `/Users/xxxxx/Documents/ReactJS/xxxxx/node_modules/react-native/package.json``

Pods installed React and seems to be causing trouble with the react-native of node_modules

Analyzing dependencies
Fetching podspec for `react-native-camera` from `../node_modules/react-native-camera`
Downloading dependencies
Using Firebase (4.8.2)
Using FirebaseAnalytics (4.0.9)
Using FirebaseAuth (4.4.2)
Using FirebaseCore (4.0.14)
Using FirebaseDatabase (4.1.4)
Using FirebaseInstanceID (2.0.8)
Using GTMSessionFetcher (1.1.13)
Using GoogleToolboxForMac (2.1.3)
Using React (0.11.0)
Using leveldb-library (1.20)
Using nanopb (0.3.8)
Using react-native-camera (1.0.0)

### Environment
    "react": "16.2.0",
    "react-native": "^0.53.0",

### react-native-camera
    "react-native-camera": "^1.0.0",

Thanks a lot.
Best regards

jgfidelis, 2018-02-17:
Please try with the master branch, a PR made recently might solve this issue

densa, 2018-02-19:
@ishigamii why you install React in pods?

closed 2018-02-19 by sibelius

ishigamii, 2018-02-20:
@densa I don't try to install react in pods, the pod line from  
`pod 'react-native-camera', path: '../node_modules/react-native-camera'` has a reference to React and install it when I launch `pod install`

@jgfidelis I changed to the master branch and did change the problem but still having problem with it. After adding to my package.json the master line, I did 
- npm install
- react-native link react-native-camera

And got the issue https://github.com/react-native-community/react-native-camera/issues/1207

dann1609, 2018-03-09:
what is the solution? I have the same problema @ishigamii

ishigamii, 2018-03-09:
There is no solution given yet and the topic was closed. @dann1609 if you have find something or If I do I will post it here

MrLoh, 2018-03-09:
@sibelius why was this closed? How should one get passed this issue, it's blocking me from upgrading to 1.0.

dann1609, 2018-03-19:
For anyone has this problem., please do a manual link:

1) npm install react-native-camera --save
2) In XCode, in the project navigator, right click Libraries ➜ Add Files to [your project's name]
3) Go to node_modules ➜ react-native-camera and add RNCamera.xcodeproj
4) In XCode, in the project navigator, select your project. Add libRNCamera.a to your project's Build Phases ➜ Link Binary With Libraries
5) Click RNCamera.xcodeproj in the project navigator and go the Build Settings tab. Make sure 'All' is toggled on (instead of 'Basic'). In the Search Paths section, look for Header Search Paths and make sure it contains both $(SRCROOT)/../../react-native/React and $(SRCROOT)/../../../React - mark both as recursive.

babourn, 2018-05-23:
It seems like the Issue is the dependency on `React` in the `react-native-camera.podspec` it's downloading a version of React (0.11.0 for some reason) into the `Pods` folder, the bundler picks up both that version of React and the one in `node_modules` installed by npm.

For a quick workaround: I added the following to `ios/Podfile`, it lets CocoaPods use the React in your top level node_modules for the `React` dependency rather than downloading a new version in Pods and fixes the Ambiguity

`  pod 'React', :path => '../node_modules/react-native', :subspecs => [
      'Core',
      'CxxBridge',
      'DevSupport',
      'RCTActionSheet',
      'RCTAnimation',
      'RCTGeolocation',
      'RCTImage',
      'RCTLinkingIOS',
      'RCTNetwork',
      'RCTSettings',
      'RCTText',
      'RCTVibration',
      'RCTWebSocket',
    ]
  pod 'yoga', :path => '../node_modules/react-native/ReactCommon/yoga'`

tc0087, 2018-07-05:
The above solution results in a problem due to CxxBridge. Does anyone have any other suggestions? Thanks!

MrLoh, 2018-07-05:
@tc0087 did you try doing the manual install, that way you can avaoid pods.