react-native-camera/react-native-camera #1239: [Android] Manifest merger failed

mistenkt, 2018-02-14:
So i have spent a day trying all suggestions for similar issues and without any luck. This also fails on a completely fresh RN project.

:react-native-camera:processDebugAndroidTestManifest FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-camera:processDebugAndroidTestManifest'.
> Manifest merger failed : Attribute meta-data#android.support.VERSION@value value=(26.0.2) from [com.android.support:exifinterface:26.0.2] AndroidManifest.xml:25:13-35
  	is also present at [com.android.support:support-v4:26.0.1] AndroidManifest.xml:28:13-35 value=(26.0.1).
  	Suggestion: add 'tools:replace="android:value"' to <meta-data> element at manifestMerger196235135338020733.xml:23:9-25:38 to override.

sibelius, 2018-02-14:
Check Q & A docs

closed 2018-02-14 by sibelius

mistenkt, 2018-02-14:
Closing this doesnt fix the issue, none of the solutions in the Q&A helps

jgfidelis, 2018-02-14:
In your app/build.gradle, change the react-native-camera compile to:
```
compile (project(':react-native-camera')) {
        exclude group: "com.android.support"
    }
compile "com.android.support:exifinterface:26.1.0"
```

mistenkt, 2018-02-14:
adding exclude to react-native-camera doesnt help. Error still shows every time on gradle sync.

jgfidelis, 2018-02-14:
```
compile (project(':react-native-camera')) {
exclude group: "com.android.support"
}
compile ("com.android.support:exifinterface:26.0.2") {
force = true;
}
```

Could you try this instead?

mistenkt, 2018-02-14:
`Error:Execution failed for task ':react-native-camera:processDebugAndroidTestManifest'.
> Manifest merger failed : Attribute meta-data#android.support.VERSION@value value=(26.0.2) from [com.android.support:exifinterface:26.0.2] AndroidManifest.xml:25:13-35
  	is also present at [com.android.support:support-v4:26.0.1] AndroidManifest.xml:28:13-35 value=(26.0.1).
  	Suggestion: add 'tools:replace="android:value"' to <meta-data> element at manifestMerger1115803340332880803.xml:23:9-25:38 to override.`

`dependencies {
    compile project(':react-native-image-picker')
    compile project(':react-native-fs')
    compile project(':react-native-fcm')
    compile 'com.google.firebase:firebase-core:11.8.0' //this decides your firebase SDK version
    compile project(':react-native-facebook-login')
    compile fileTree(dir: "libs", include: ["*.jar"])
    compile "com.android.support:appcompat-v7:26.0.2"
    compile "com.facebook.react:react-native:+"  // From node_modules
    compile (project(':react-native-camera')) {
        exclude group: "com.google.android.gms"
    }
    compile ("com.android.support:exifinterface:26.0.2") {
        force = true;
    }
    compile ("com.google.android.gms:play-services-vision:11.8.0") {
        force = true;
    }
}`

jgfidelis, 2018-02-14:
@mistenkt 

Please, look at your build.gradle:
```
compile (project(':react-native-camera')) { 
exclude group: "com.google.android.gms" 
}
```
It should be:
```
compile (project(':react-native-camera')) { 
exclude group: "com.google.android.gms" 
exclude group: "com.android.support"
}
```

mistenkt, 2018-02-14:
tried them all. updated now to be both and same error

mistenkt, 2018-02-14:
it is now:

dependencies {
    compile project(':react-native-image-picker')
    compile project(':react-native-fs')
    compile project(':react-native-fcm')
    compile 'com.google.firebase:firebase-core:11.8.0' //this decides your firebase SDK version
    compile project(':react-native-facebook-login')
    compile fileTree(dir: "libs", include: ["*.jar"])
    compile "com.android.support:appcompat-v7:26.0.2"
    compile "com.facebook.react:react-native:+"  // From node_modules
    compile (project(':react-native-camera')) {
        exclude group: "com.android.support"
        exclude group: "com.google.android.gms"
    }
    compile ("com.android.support:exifinterface:26.0.2") {
        force = true;
    }
    compile ("com.google.android.gms:play-services-vision:11.8.0") {
        force = true;
    }
}

and it gives them same error for Manifest merger

sibelius, 2018-02-14:
follow this https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md#meta-data-android-26

mistenkt, 2018-02-14:
i have

sibelius, 2018-02-14:
can u create a repo to reproduce this error?

try to run this one https://github.com/react-native-community/rncamera-example

jgfidelis, 2018-02-14:
Ok that is pretty weird.

remove 
```
compile ("com.android.support:exifinterface:26.0.2") {
force = true;
}
```

It should build, try to take a photo and see if everything goes ok.

matheus-lima, 2018-02-21:
this worked for me.
adding a meta-data to your manifest

```xml
<meta-data
        tools:replace="android:value"
        android:name="android.support.VERSION"
        android:value="26.0.2" />
```

afilp, 2018-02-21:
@matheus-lima @sibelius Does this mean that a new version could be released that will fix all these problems?