react-native-camera/react-native-camera #1246: Torch is deactivated on record start

ma-pe, 2018-02-14:
### Which implementation are you using

*RNCamera*

### Steps to reproduce
1. Use camera with torch activated
2. Start video recording

### Expected behaviour
The torch stays activated. 

### Actual behaviour
The torch is switched off, when recording is started.

### Environment
- **React Native version**: 0.53.0
- **React Native platform + platform version**: iOS 11.2.5 (Android works as expected)

### react-native-camera
**Version**: "master"

jgfidelis, 2018-02-15:
this is the second bug related to ios torch (also #1189).

Can you test if it happens with the expo camera component as well?

use this app: https://expo.io/@community/camerja

ma-pe, 2018-02-15:
I sure would, but you can't record videos (only photos) with the expo-test-app. :(

jgfidelis, 2018-02-15:
Sorry haha. I will take a look into it eventually.

imspikey, 2018-02-27:
I have the same problem.
also I have used  a react-native-flash for turning flashlight manually and whenever camera starts recording flash turns of instantly ???!!
Please can you fix this issue 

Environment
React Native version: 0.53.0
React Native platform + platform version: iOS 11.2.5 (Android works as expected)
"react": "16.2.0",
 "react-native": "^0.53.3",
"react-native-camera": "^1.0.1",

sibelius, 2018-03-14:
does it make sense to incorporate this https://github.com/KrowdBeat/react-native-flash

in this repo?

closed 2018-03-23 by jgfidelis

edi, 2020-04-25:
I'm posting about this everywhere ... but it would be great if `react-native-flash` gets added to Expo ... 

There are so many use cases for it. One of them is that we need a toggle for the camera's flash when you want to scan a barcode during the night.

Another one is morse code communication between two devices at night, when there's no connection between a couple of households.

Please advise.