react-native-camera/react-native-camera #1248: Undefined is not an object (evaluating 'CameraManager.Aspect')

patlux, 2018-02-15:
Hey,

Yes, another thread about this issue:

- https://github.com/react-native-community/react-native-camera/issues/1207
- https://github.com/react-native-community/react-native-camera/issues/896
- https://github.com/react-native-community/react-native-camera/issues/877
- https://github.com/react-native-community/react-native-camera/issues/748
- and so on...

### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

I did that. And I created 3 another projects from scratch with `react-native init`. (v0.53.0, v0.52.0, v0.52.2).

### Which implementation are you using

*RNCamera* or RCTCamera (RCTCamera will be removed on v2.0.0)?

Tried with both.

### Steps to reproduce
1. `react-native init <projectname> --version=0.53.0`
and/or
`react-native init <projectname> --version=0.52.2`
and/or
`react-native init <projectname> --version=0.51.0`
2. cd <projectname>
3. yarn add react-native-camera
4. cd ios
5. pod init
6. cd ..
7. react-native link react-native-camera
8. Build success, but runtime error with this issue.

### Expected behaviour

It should work.

### Actual behaviour

Runtime error:

[image]

### Environment
- **Node.js version**: v9.3.0
- **React Native version**: v0.51.0, v0.52.2, v0.53.0
- **React Native platform + platform version**: iOS 11.0

### react-native-camera
**Version**: npm version or "master" --> Tried with latest v1.0.1 and master.

Output of `console.log(NativeModules)`:

[image]

My Podfile:

```
source 'https://github.com/CocoaPods/Specs.git'

platform :ios, '8.0'
project '<projectname>.xcodeproj'

target '<projectname>' do
  pod 'yoga', :path => '../node_modules/react-native/ReactCommon/yoga'
  pod 'React', path: '../node_modules/react-native', :subspecs => [
    'Core',
    'DevSupport', # Include this to enable In-App Devmenu if RN >= 0.43
    'BatchedBridge'
  ]

  pod 'react-native-camera', path: '../node_modules/react-native-camera'
end

``` 

What I'm doing wrong? As a last step I cloned this repo and tried it with the `Example` project. But it also not working...

Thanks for any help...

sibelius, 2018-02-15:
the link is wrong

can u try to use `react-native link` instead of `cocoapods`?

nkov, 2018-02-15:
+1. Can confirm that the issue is present with both `react-native link` and `cocoapods`.

nkov, 2018-02-16:
@patlux It worked for me to remove the previously-installed `react-native-camera` pod:
```
$ rm -rf ios/Pods
$ pod install
```
and then manually link it as described [here in the readme](https://github.com/react-native-community/react-native-camera#manual-install). Then it wouldn't build until I also deleted the `FaceDetection` folder from the linked library as mentioned in the readme, now all is well.

sibelius, 2018-02-16:
Tks @nkov

closed 2018-02-16 by sibelius

bkvirendra, 2018-02-16:
@nkov I have an react-native integration in an existing app. I tried following your instructions above (by deleting the `Pods` dir). But instead of installing manually, I have the pods install for the lib. But still getting the same error, any way to get this working with cocoapods instead of manual linking?

patlux, 2018-02-16:
I completely removed cocoapods from my project and now its working with a simply `react-native link react-native-camera`. @bkvirendra 

I don't know why it's not working with cocoapods.

bkvirendra, 2018-02-16:
@patlux I found the issue in the `podspec` file. The `source_files` attribute in podspec wasn't in sync with the current directory structure of the project. Created a PR here -> https://github.com/react-native-community/react-native-camera/pull/1256

arapocket, 2018-03-06:
@bkvirendra Where is this podspec file exactly?

bkvirendra, 2018-03-06:
@arapocket The podspec file in the project is `react-native-camera.podspec`.

Can you try installing using the current master branch instead (since as of now, the patch is not included in a release), for eg. adding the following in your `package.json` 
```
"react-native-camera": "https://github.com/react-native-community/react-native-camera/tarball/master"
```
Also in your Xcode, in the project viewer check whether the Pods project includes files in `RCT`, `RN` directories. The files should be the same RCTCamera.h RCTCamera.m etc..?

arapocket, 2018-03-07:
@bkvirendra All of these issues are from `react-native link`. I reinstalled without using that and everything worked fine.

And to clarify, using `react-native link` and then trying the manual install won't work. You need to start fresh.

7772, 2018-03-09:
@nkov's method does work in my case! Thanks 👍

pohsiu, 2018-03-09:
I use the same way with @nkov 
I just remove the "FaceDetection folder" from Library and manual link again then it works

fazeelzama20, 2018-06-19:
worked by using `cocoapod` , just use command `cd ios` -> `pod install` then `react-native run-ios`

Jont828, 2018-06-19:
I'm using Expo and CRNA. I tried removing the FaceDetection folder, but it's still not working. I'm also using `react-native link` since I'm not really familiar on how to do things manually, yet. Should I switch to `cocoapod` instead?

7772, 2018-06-19:
@Jont828 

You can not use Expo and CRNA if you used native modules(ex.. camera). According to Expo document, You should eject your project made of Expor or CRNA. 
But, I recommend that you make your project using "react-native init" from the beginning because you can not use Expo and CRNA after you did eject your project.
In my case, The more I experience projects, the more I am reluctant to use Expo and CRNA.

riyadshauk, 2019-02-01:
I had a similar issue just now. My solution? Just use Camera from the Expo SDK (since I was already using Expo and not yet detached): [https://docs.expo.io/versions/v32.0.0/sdk/camera/](https://docs.expo.io/versions/v32.0.0/sdk/camera/). At least their provided example code runs fine... will see how further features integrations play out using that.