react-native-camera/react-native-camera #1249: Black flashing screen when recording video (also saved in the recording)

n1ru4l, 2018-02-15:
### Which implementation are you using

*RNCamera*

### Steps to reproduce

1. Mount a video component
2. Use the reference prop to start recording when pressing the screen

### Expected behaviour
The video start recording without the black going screen for a few milliseconds. Also the video should be saved without black frames at the beginning

### Actual behaviour
The screen turns black for a few milliseconds. The few milliseconds of dark frames are also present in the recording.

### Environment
- **Node.js version**: v8.9.4
- **React Native version**: v0.52.2
- **React Native platform + platform version**: iOS 10

### react-native-camera
**Version**: v1.0.1

n1ru4l, 2018-02-26:
Duplicate of https://github.com/expo/expo/issues/1057

n1ru4l, 2018-05-02:
@jgfidelis @sibelius Would you accept a PR on `RNCamera` that implements a property for the current mode (`recording` and `picture`), in order to prevent the flashing when recording?

Also there must be a solution to this problem, bot snapchat and instagram can handle this, so it should not be impossible. We might get some inspiration (or even reuse?) some open source solution which is written in objective-c?

sibelius, 2018-05-02:
We will accept a PR to fix this

n1ru4l, 2018-05-02:
@delta98 Any ideas what snapchat/instagram is doing differently? Because from what I can see they are also switching between the modes without flashings

delta98, 2018-05-04:
@n1ru4l it looks like it happens after calling `[self setupMovieFileCapture];`

delta98, 2018-05-04:
Actually nailed this down even further, when `record` is called, the `session` configuration is changed, this causes the current session to be unlocked, which causes the flicker. 

`updateSessionPreset` and `updateSessionAudioIsMuted` use this within their function calls. 

Removing these calls from record and moving `setupMovieFileCapture` to `initializeCaptureSessionInput` or `initWithBridge` allows capture to active more _seamless_. 

I'll put a PR together which takes into account the above and provides the same functionality, but likely will need to limit the camera mode to an instance of camera.

delta98, 2018-05-04:
@n1ru4l could you checkout this branch and confirm whether this stops the flicker you see https://github.com/react-native-community/react-native-camera/pull/1542

Will also need to soften some of the changes and re-introduce functionality for face detection.

delta98, 2018-05-04:
@sibelius ^^

n1ru4l, 2018-05-06:
@delta98 Will try to do it tomorrow

watadarkstar, 2018-05-29:
Following! Will test this out. Hows the PR going @n1ru4l ?

n1ru4l, 2018-05-29:
@watadarkstar I am using it in production 😉

pqminh, 2018-08-20:
How to fix it?

n1ru4l, 2018-08-20:
@pqminh my current workaround is described here: https://github.com/react-native-community/react-native-camera/pull/1542#issuecomment-407324574

If you want to use the newest master you will have to adjust it.

I still had no time to file in a pull request 😕

pqminh, 2018-08-21:
@n1ru4l tks. I'm trying now.

listicos, 2018-10-25:
Hello, I just created a simple solution for this https://github.com/react-native-community/react-native-camera/pull/1886

n1ru4l, 2018-10-29:
Fixed in #1886

closed 2018-10-29 by n1ru4l

Thanhcao86, 2019-03-29:
I tried but still encountered this situation again in the second video recording, I thought because of some method called back wrongly, I tried to read and cancel some unnecessary methods for my project but didn't work effective.
I tried again, I found out that during the second video recording, the camera's status was again blinking, I tried to delete this method [CleanupCamera], it no longer flashes but after the second time it has an exception with "Video is not recorded". I am still looking for a solution, hoping to help this situation.