react-native-camera/react-native-camera #1250: Argument list too long: recursive header expansion

nbennink, 2018-02-15:
### Steps to reproduce
1. Follow the "Manual install" guide for iOS in the readme.md file
2. Build the app

### Expected behaviour
Builds the app and use the package.

### Actual behaviour
Throws an error when building:
`
Argument list too long: recursive header expansion failed at /Users/nbennink/Documents/projects/react-native-app/node_modules/react-native-camera/ios/../../../ios/build/Index/DataStore/v5/records/JR.
`
I don't get the app working. I also tried doing the "mostly automatic install" but then I get the error: Cannot read property Aspect from undefined (which seems to be an issue with linking the package).

I tried it a several times. The app builds correctly when removing the react-native-camera package

### Environment
- **Node.js version**: V9.4.0
- **React Native version**: V0.52.1
- **React Native platform + platform version**: iOS 11

### react-native-camera
**Version**: V1.0.1

jgfidelis, 2018-02-15:
Are you trying to build with Face detection or not?

nbennink, 2018-02-15:
@jgfidelis no, I'm not using face detection.

jgfidelis, 2018-02-15:
did you remove the FN folder correctly and cleaned the project before building?

nbennink, 2018-02-16:
@jgfidelis Yes I have cleaned the project before building. I also tried to build in on device and on emulator but it doesn't make a difference.

What is the FN folder and where can I remove it?

jgfidelis, 2018-02-16:
Please follow the section "Face Detection Steps" in the readme

closed 2018-02-16 by jgfidelis

kpmil, 2018-02-23:
Hi, I followed the instructions given in the readme, but end up having a similar issue - throws an error when cleaning project:

`Argument list too long: recursive header expansion failed at /Users/kpmil/react-native-app/node_modules/react-native-camera/ios/../../../ios/Pods/React/node_modules/browser-resolve/node_modules/resolve/test/resolver/other_path`

- Node.js version 9.5.0
- React Native version 0.50.3
- React Native platform + platform version iOS 10
- react-native-camera version 1.0.1

kpmil, 2018-02-28:
Recursive header expansion issue seems to be fixed, thx @MatthieuLemoine. 

Now the compilation fails with a linker error, though:

`ld: warning: directory not found for option '-L/Users/kpmil/Library/Developer/Xcode/DerivedData/Build/Products/Debug-iphonesimulator/react-native-camera

ld: library not found for -lreact-native-camera

clang: error: linker command failed with exit code 1 (use -v to see invocation)`

nbennink, 2018-02-28:
@MatthieuLemoine 's pull request fixed it for me! Thanks 🥇

kpmil, 2018-03-01:
that fixed the issue for me, thx @MatthieuLemoine

avinashwigroup, 2018-04-02:
I'm still getting this issue. It showed up after installing react-native-maps. The react-native-camera version is 1.0.3

MaheshNandam, 2018-04-05:
I have added Pods at frame relative paths and also [replace master repo at package.json](https://github.com/react-native-community/react-native-camera#how-to-use-master-branch)

but still:
Argument list too long: recursive header expansion failed at ../node_modules/react-native-camera/ios/../../../ios/Pods/React/node_modules/fsevents/node_modules/path-is-absolute.

@lwansbrough @jgfidelis, can anyone guide to solve this issue?

Thanks a lot!.

jgfidelis, 2018-04-05:
@avinashwigroup can you try master branch? Clean your project and restart xcode.

avinashwigroup, 2018-04-05:
@jgfidelis : The issue is now solved. react-native link ... didn't link it correctly, so I did it manually.

thaoth58, 2018-04-09:
Sorry @avinashwigroup, can you tell me how to fix this issue. I have just followed the instruction but it doesn't work.

avinashwigroup, 2018-04-09:
@thaoth58 : Try to install it manually rather than using the 'react-native link' method.  The issue with mine was there were duplicates libs linked/included inside the android project and I had to remove them.  Also, the order in which you add them inside your app/build.gradle file is very important; put it last.

thaoth58, 2018-04-09:
@avinashwigroup : It works well on Android, but on iOS i got the "...Argument list too long" issue.

rehabaam, 2018-04-12:
I have read all the issues related to this and done all the steps as suggested but none worked! Even, I unlinked the package, removed it, added the master repo, linked again, added Pod and did all the steps again but It is still giving this "...Argument list too long" issue. 
Any solution?

MaheshNandam, 2018-04-12:
hey @rehabaam, 
Did you try removing 'react-native-camera' from pod file?

rehabaam, 2018-04-18:
@MaheshNandam, yes, I did but that didn't fix the issue. 
After two days of checking and testing, I cleaned up the project and tried the manual installation only where I edited the file "project.pbxproj" and removed **all** the entries for "SEARCH_PATHS" and then did build the project using xCode which was done successfully. After that, I added the HEADER_SEARCH_PATHS entries; now everything works fine!

woodpav, 2018-06-28:
#1407 #1648 #1407 

I think we getting the same error for different reasons. My solution was to change `$(SRCROOT)/../../../ios` to `$(SRCROOT)/../../../../ios`. I fixed it by reading my error message and checking the paths. 

Best of luck!

ducpt2, 2018-07-23:
Thanks @rehabaam .
 Follow his solution will work.

platonish, 2018-07-24:
@rehabaam Thanks, your solution worked. I had to first clear both FRAMEWORK_SEARCH_PATHS and HEADER_SEARCH_PATHS --> build (builds successfully)--> add the HEADER_SEARCH_PATHS mentioned in the readme --> build again.