react-native-camera/react-native-camera #1251: RNCamera is uninstalled when you install another npm package...... ☠️

emmajam, 2018-02-16:
### Which implementation are you using

*RNCamera*

### Steps to reproduce
1. `npm install react-native-camera --save`
2. `cd node_modules && ls | grep react-native-camera` - confirm its there ☑️ 
2. `npm install react-native-XXXX --save` install a different package
3. `cd node_modules && ls | grep react-native-camera` -- its gone

### Expected behaviour
`react-native-camera` should only be uninstalled if `npm uninstall xxxx` has been run. It should not be removed when you install another package after it 😢 

### Actual behaviour
Install react-native-camera then install another react-native package - you will see rn-camera has been removed from node modules.

### Environment
- **Node.js version**:`v9.2.1`
- **React Native version**: `react-native-cli: 2.0.1 react-native: 0.51.0`
- **NPM version**: `5.6.0`
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: "master"

sibelius, 2018-02-16:
Maybe a problem with npm

closed 2018-02-16 by sibelius

v4lerio, 2018-08-06:
I still have the same issue. No idea why it is happening