react-native-camera/react-native-camera #1252: "Could not resolve all dependencies for configuration ':app:_debugApkCopy'"

FOLLGAD, 2018-02-16:
### Which implementation are you using
Was using pre-1.0.0 without problems. When updating and migrating I started recieved errors.

### Expected behaviour
When doing react-native run-android inside the project folder it should start the app on the emulated device.
It worked fine until I tried updating and migrating to 1.0.0. I edited the files according to migration.md.

### Actual behaviour
I get an error:
```
A problem occurred configuring project ':app'.
> Could not resolve all dependencies for configuration ':app:_debugApkCopy'.
   > Could not find com.android.support:support-v4:26.0.1.
     Searched in the following locations:
         file:/C:/AndroidSDK/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.pom
         file:/C:/AndroidSDK/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
         file:/C:/Users/Emil/Documents/GitHub/InventoryApp/android/sdk-manager/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
     Required by:
         InventoryApp:app:unspecified > com.android.support:appcompat-v7:23.0.1
   > Could not find com.android.support:support-annotations:26.0.1.
     Searched in the following locations:
         file:/C:/AndroidSDK/extras/android/m2repository/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.pom
         file:/C:/AndroidSDK/extras/android/m2repository/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.jar
         file:/C:/Users/Emil/Documents/GitHub/InventoryApp/android/sdk-manager/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.jar
     Required by:
         InventoryApp:app:unspecified > InventoryApp:react-native-camera:unspecified > com.android.support:exifinterface:26.0.0-alpha1
   > Could not find com.android.support:support-annotations:26.0.1.
     Searched in the following locations:
         file:/C:/AndroidSDK/extras/android/m2repository/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.pom
         file:/C:/AndroidSDK/extras/android/m2repository/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.jar
         file:/C:/Users/Emil/Documents/GitHub/InventoryApp/android/sdk-manager/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.jar
     Required by:
         InventoryApp:app:unspecified > InventoryApp:react-native-camera:unspecified > com.github.react-native-community:cameraview:d5d9b0d925494ef451ce3eef3fdf14cc874d9baa
   > Could not find com.android.support:support-v4:26.0.1.
     Searched in the following locations:
         file:/C:/AndroidSDK/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.pom
         file:/C:/AndroidSDK/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
         file:/C:/Users/Emil/Documents/GitHub/InventoryApp/android/sdk-manager/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
     Required by:
         InventoryApp:app:unspecified > InventoryApp:react-native-camera:unspecified > com.github.react-native-community:cameraview:d5d9b0d925494ef451ce3eef3fdf14cc874d9baa
   > Could not find com.android.support:support-v4:26.0.1.
     Searched in the following locations:
         file:/C:/AndroidSDK/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.pom
         file:/C:/AndroidSDK/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
         file:/C:/Users/Emil/Documents/GitHub/InventoryApp/android/sdk-manager/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
     Required by:
         InventoryApp:app:unspecified > com.google.android.gms:play-services-vision:10.2.0 > com.google.android.gms:play-services-basement:10.2.0
```
I have tried on two computers, and I get the same result.

### Environment
- **Node.js version**: 9.1.0
- **React Native version**: 0.52
- **React Native platform + platform version**: Android 6.0, etc

### react-native-camera
**Version**: 1.0.1

sibelius, 2018-02-16:
Use latest gradle, android studio could fix this as well

Try to add

google() to maven repositories

closed 2018-02-16 by sibelius

rohitpaniker, 2018-02-23:
Did that, didn't work:

* What went wrong:
A problem occurred configuring project ':app'.
> Could not resolve all dependencies for configuration ':app:_debugApkCopy'.
   > Could not find com.android.support:support-v4:26.0.1.
     Searched in the following locations:
         file:/C:/Andoid/sdk/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.pom
         file:/C:/Andoid/sdk/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
         file:/D:/ROHIT/Projects/ClassDalal/cdalal/android/sdk-manager/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
     Required by:
         cdalal:app:unspecified > com.android.support:appcompat-v7:23.0.1
         cdalal:app:unspecified > com.facebook.react:react-native:0.20.1 > com.android.support:recyclerview-v7:23.0.1
   > Could not find com.android.support:support-annotations:26.0.1.
     Searched in the following locations:
         file:/C:/Andoid/sdk/extras/android/m2repository/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.pom
         file:/C:/Andoid/sdk/extras/android/m2repository/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.jar
         file:/D:/ROHIT/Projects/ClassDalal/cdalal/android/sdk-manager/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.jar
     Required by:
         cdalal:app:unspecified > cdalal:react-native-camera:unspecified > com.android.support:exifinterface:26.0.0-alpha1
   > Could not find com.android.support:support-annotations:26.0.1.
     Searched in the following locations:
         file:/C:/Andoid/sdk/extras/android/m2repository/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.pom
         file:/C:/Andoid/sdk/extras/android/m2repository/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.jar
         file:/D:/ROHIT/Projects/ClassDalal/cdalal/android/sdk-manager/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.jar
     Required by:
         cdalal:app:unspecified > cdalal:react-native-camera:unspecified > com.github.react-native-community:cameraview:d5d9b0d925494ef451ce3eef3fdf14cc874d9baa
   > Could not find com.android.support:support-v4:26.0.1.
     Searched in the following locations:
         file:/C:/Andoid/sdk/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.pom
         file:/C:/Andoid/sdk/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
         file:/D:/ROHIT/Projects/ClassDalal/cdalal/android/sdk-manager/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
     Required by:
         cdalal:app:unspecified > cdalal:react-native-camera:unspecified > com.github.react-native-community:cameraview:d5d9b0d925494ef451ce3eef3fdf14cc874d9baa
   > Could not find com.android.support:support-annotations:26.0.1.
     Searched in the following locations:
         file:/C:/Andoid/sdk/extras/android/m2repository/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.pom
         file:/C:/Andoid/sdk/extras/android/m2repository/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.jar
         file:/D:/ROHIT/Projects/ClassDalal/cdalal/android/sdk-manager/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.jar
     Required by:
         cdalal:app:unspecified > com.facebook.react:react-native:0.20.1 > com.android.support:recyclerview-v7:23.0.1
   > Could not find com.android.support:support-v4:26.0.1.
     Searched in the following locations:
         file:/C:/Andoid/sdk/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.pom
         file:/C:/Andoid/sdk/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
         file:/D:/ROHIT/Projects/ClassDalal/cdalal/android/sdk-manager/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
     Required by:
         cdalal:app:unspecified > com.facebook.react:react-native:0.20.1 > com.facebook.fresco:fresco:0.8.1 > com.facebook.fresco:imagepipeline:0.8.1
         cdalal:app:unspecified > com.facebook.react:react-native:0.20.1 > com.facebook.fresco:fresco:0.8.1 > com.facebook.fresco:drawee:0.8.1
   > Could not find com.android.support:support-v4:26.0.1.
     Searched in the following locations:
         file:/C:/Andoid/sdk/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.pom
         file:/C:/Andoid/sdk/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
         file:/D:/ROHIT/Projects/ClassDalal/cdalal/android/sdk-manager/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
     Required by:
         cdalal:app:unspecified > cdalal:react-native-camera:unspecified > com.google.android.gms:play-services-vision:8.4.0 > com.google.android.gms:play-services-base:8.4.0 > com.google.android.gms:play-services-basement:8.4.0

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 45.725 secs
Could not install the app on the device, read the error above for details.
Make sure you have an Android emulator running or a device connected and have
set up your Android development environment:
https://facebook.github.io/react-native/docs/getting-started.html

sibelius, 2018-02-23:
missing android support lib

rohitpaniker, 2018-02-23:
I did everything, from doing as per what your ToDo doc says to updating SDK via SDK Manager, re-Installing latest Android Studio IDE (3.0.1) along with latest Gradle but the error doesn't seem to go away or resolve.

slimansaid, 2018-02-23:
Same issue here.

FOLLGAD, 2018-02-23:
@rohitpaniker @slimansaid What I did was to go into Android Studio and open the "android" directory in my project, then letting it fix the errors it found. Maybe it works for you too?

rohitpaniker, 2018-02-24:
@FOLLGAD  Alright, this worked for me.

Now I am facing the below errors:

Building and installing the app on the device (cd android && gradlew.bat installDebug)...

> Configure project :app
Configuration 'compile' in project ':app' is deprecated. Use 'implementation' instead.

> Configure project :react-native-camera
Configuration 'compile' in project ':react-native-camera' is deprecated. Use 'implementation' instead.

> Task :react-native-camera:compileDebugJavaWithJavac
D:\ROHIT\Projects\ClassDalal\cdalal\node_modules\react-native-camera\android\src\main\java\org\reactnative\camera\CameraModule.java:14: error: cannot find symbol
import com.facebook.react.uimanager.UIBlock;
                                   ^
  symbol:   class UIBlock
  location: package com.facebook.react.uimanager
D:\ROHIT\Projects\ClassDalal\cdalal\node_modules\react-native-camera\android\src\main\java\org\reactnative\camera\CameraModule.java:191: error: cannot find symbol
    uiManager.addUIBlock(new UIBlock() {
                             ^
  symbol:   class UIBlock
  location: class CameraModule
D:\ROHIT\Projects\ClassDalal\cdalal\node_modules\react-native-camera\android\src\main\java\org\reactnative\camera\CameraModule.java:221: error: cannot find symbol
      uiManager.addUIBlock(new UIBlock() {
                               ^
  symbol:   class UIBlock
  location: class CameraModule
D:\ROHIT\Projects\ClassDalal\cdalal\node_modules\react-native-camera\android\src\main\java\org\reactnative\camera\CameraModule.java:244: error: cannot find symbol
      uiManager.addUIBlock(new UIBlock() {
                               ^
  symbol:   class UIBlock
  location: class CameraModule
D:\ROHIT\Projects\ClassDalal\cdalal\node_modules\react-native-camera\android\src\main\java\org\reactnative\camera\CameraModule.java:265: error: cannot find symbol
      uiManager.addUIBlock(new UIBlock() {
                               ^
  symbol:   class UIBlock
  location: class CameraModule
D:\ROHIT\Projects\ClassDalal\cdalal\node_modules\react-native-camera\android\src\main\java\org\reactnative\camera\events\BarCodeReadEvent.java:32: error: method init in class Event<T> cannot be applied to given types;
    super.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
D:\ROHIT\Projects\ClassDalal\cdalal\node_modules\react-native-camera\android\src\main\java\org\reactnative\camera\events\CameraMountErrorEvent.java:22: error: method init in class Event<T> cannot be applied to given types;
    event.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
D:\ROHIT\Projects\ClassDalal\cdalal\node_modules\react-native-camera\android\src\main\java\org\reactnative\camera\events\CameraReadyEvent.java:22: error: method init in class Event<T> cannot be applied to given types;
    event.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
D:\ROHIT\Projects\ClassDalal\cdalal\node_modules\react-native-camera\android\src\main\java\org\reactnative\camera\events\FaceDetectionErrorEvent.java:24: error: no suitable method found for init(int)
    event.init(viewTag);
         ^
    method Event.init(int,long) is not applicable
      (actual and formal argument lists differ in length)
    method FaceDetectionErrorEvent.init(int,RNFaceDetector) is not applicable
      (actual and formal argument lists differ in length)
D:\ROHIT\Projects\ClassDalal\cdalal\node_modules\react-native-camera\android\src\main\java\org\reactnative\camera\events\FaceDetectionErrorEvent.java:29: error: method init in class Event<T> cannot be applied to given types;
    super.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
D:\ROHIT\Projects\ClassDalal\cdalal\node_modules\react-native-camera\android\src\main\java\org\reactnative\camera\events\FacesDetectedEvent.java:52: error: method init in class Event<T> cannot be applied to given types;
    super.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
11 errors

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-camera:compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

JAVA_HOME and ANDROID_HOME are all properly placed too in environment variables

sibelius, 2018-02-24:
It is getting the wrong React native version

rohitpaniker, 2018-02-24:
Resolved it. Steps which worked for me:

1) Downloaded latest Android Studio uninstalled old one and installed this new downloaded version
2) Deleted project created using react-native init projectName
3) Recreated the project using above command and installed all dependencies (yarn install)
4) Opened projectName/android in Android Studio
5) Made changes in files as per the instructions defined in https://github.com/react-native-community/react-native-camera
6) Made the below changes in build.gradle(android/app/):

    compileSdkVersion 26
    buildToolsVersion "26.0.2"

    defaultConfig {
        applicationId "com.projectname"
        minSdkVersion 16
        targetSdkVersion 26
}

compile "com.android.support:appcompat-v7:26.0.2" // this version should match with buildToolsVersion

7) Android studio would ask if you want to upgrade Gradle version and Gradle plugin version, click on Yes/Update

8) At cmd prompt, executed: react-native run-android

*PS: If you see "Cannot delete path" error then run : cd android && gradlew clean and then re-run previous command

slimansaid, 2018-02-24:
this worked for me while opening Android Studio to resolve the dependencies

`allprojects {
    repositories {
        mavenLocal()
        jcenter()
        maven {
            url 'https://maven.google.com'
        }
        maven {
            url "https://jitpack.io"
        }
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
    }
}`

kristerkari, 2018-03-14:
The config from @GoekhanPolat was enough to fix the error for me 👍

luojinghui, 2018-03-21:
请保持能够访问到外网！Anything can be solved.

tobiaskusnaman, 2018-03-25:
Thanks @xZliman it works !

NguyenPhatBG, 2018-06-01:
step 01: gradle-wrapper.properties change to : https\://services.gradle.org/distributions/gradle-4.4-all.zip
step 02: root build.gradle change to:  classpath 'com.android.tools.build:gradle:3.1.0'
step03: app/../build.gradle 
android {
    compileSdkVersion 27

    defaultConfig {
        applicationId "com.rn_pjsip"
        minSdkVersion 21
        targetSdkVersion 27
        versionCode 1
        versionName "1.0"
        ndk {
            abiFilters "armeabi-v7a", "x86"
        }
    }
...

kaushal9678, 2018-08-30:
**update your project build.gradle dependencies
 dependencies {
        classpath 'com.android.tools.build:gradle:3.1.4'
        classpath 'com.google.gms:google-services:3.2.1'
}**

anubhav19rai, 2018-09-12:
Please run this command **react-native upgrade**. This command will ask replace existing file or not. If you    want to replace a particular file press Y and hit enter button or press N. After completing this command than run command  **react-native link**. This command will link all dependencies than you can run command **react native run-android /ios**. This issue create when you uninstall some dependencies so sometimes  it do not update build files like setting.gradle build.gradle etc.

kaushal9678, 2018-09-12:
@anubhav19rai  some dependencies are not added using the **react-native link** you have to add those manually on both platforms. Try manually!

benhartouz, 2018-10-26:
```
allprojects {
    repositories {
        mavenLocal()
        google()
        maven {
            url 'https://maven.google.com/'
            name 'Google'
        }
        maven {
            url 'https://jitpack.io'
        }
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
        jcenter()
        
    }
}
```

**this code work for me very well**  just put jcenter() function after google() function.

galcivar, 2018-11-24:
Besides reinstalling everything, nothing else has worked...added repositories, changed sdk to:
        buildToolsVersion = "27.0.3"
        minSdkVersion = 16
        compileSdkVersion = 27
        targetSdkVersion = 27
        supportLibVersion = "27.1.1"

Tried every single other version but I get that gradle uses 27.0.3 by default now

JesseScott, 2019-09-20:
you shouldn't need both of

`       google()`
and
```
        maven {
            url 'https://maven.google.com/'
            name 'Google'
        }
```

azim254, 2020-03-30:
add this 

 maven { url 'https://maven.google.com' }