react-native-camera/react-native-camera #1254: Image not saved to Camera Roll with new RNCamera?

donni106, 2018-02-16:
### Which implementation are you using

*RNCamera*

I updated my application to master with RNCamera. The app is running in an iOS Simulator.

### Steps to reproduce
Take picture with new `takePictureAsync()`.

### Behaviour
With old version RCTCamera the image was stored in the devices Camera Roll right after capturing.
Now i capture and get the data. I can output it with `console.log` and see it in the desktop browser when clicking the link.
How can i achieve this with the new version RNCamera?

### Environment
- **Node.js version**: v9.5.0
- **React Native version**: 0.51.0
- **React version**: 16.0.0

### react-native-camera
**Version**: "master" 1.0.1

### Something else
With checking the new docs, i saw
```
toggleView() {
    this.setState({
      showGallery: !this.state.showGallery,
    });
  }
```

and implemented it on another button next to my takePicture. But nothing happens. What should this action actually do?

sibelius, 2018-02-16:
I think this is a feature in fact, it saves all images to a cache folder

then you can use `react-native-fs` to move to somewhere else

jgfidelis, 2018-02-16:
It saves the picture in your device's cache folder and the promise will return its URI. You can then move that file to anywhere you'd like with other libs like `react-native-fs`, like @sibelius mentioned.

Save directly to camera roll would be a nice feature to have that can be implemented in the future, but in RNCamera it is not available yet, only on RCTCamera.

jgfidelis, 2018-02-16:
About the `Something else`, it is a code snippet from the example app and it doesn't do anything, the variable showGallery in the state is not used for anything if I recall correctly

closed 2018-02-16 by jgfidelis

sibelius, 2018-02-16:
maybe we should add this do `docs`

Friendly-Robot, 2018-04-19:
Found this solution in a similar issue:
Use `CameraRoll.saveToCameraRoll( data.uri )` to save directly to the native photo/video library. Beats installing 'react-native-fs' if your app doesn't require any extra juggling.

CescCesc, 2018-06-16:
I'm try CameraRoll.saveToCameraRoll(data.uri, "photo") and success. 
data = await this.camera.takePictureAsync(options) in RN Camera

sundariragavan, 2018-10-31:
return await CameraRoll.saveToCameraRoll(uri, "photo");
In android, app is save the image into camera gallery. How to create a custom folder and save the image into gallery.

Friendly-Robot, 2018-12-09:
@ozankonur Try console logging the `data` response before you attempt to access `data.uri`. Your options sets `base64: true` so maybe the uri is unavailable whereas `data.base64` is.

ozankonur, 2018-12-13:
@Friendly-Robot  Thanks it helped me.

TheAlgo, 2019-02-07:
@ozankonur what did you do to save the image in the gallery?

ozankonur, 2019-02-07:
@TheAlgo I use above my codes & I removed 'base64: true' from options.

TheAlgo, 2019-02-07:
@ozankaur did this save the image to gallery ?

ozankonur, 2019-02-07:
@TheAlgo yes

TheAlgo, 2019-02-07:
[image]
@ozankonur  I am getting this and my image is not getting saved

TheAlgo, 2019-02-07:
undefined is not an object data.uri , it is showing @ozankonur