react-native-camera/react-native-camera #1255: java.lang.IllegalArgumentException: Invalid image data size.

sibelius, 2018-02-16:
```
Caused by java.lang.IllegalArgumentException: Invalid image data size.
       at com.google.android.gms.vision.Frame$Builder.setImageData(Unknown Source)
       at org.reactnative.facedetector.RNFrameFactory.buildFrame(RNFrameFactory.java:16)
       at org.reactnative.camera.tasks.FaceDetectorAsyncTask.doInBackground(FaceDetectorAsyncTask.java:40)
       at org.reactnative.camera.tasks.FaceDetectorAsyncTask.doInBackground(FaceDetectorAsyncTask.java:10)
       at android.os.AsyncTask$2.call(AsyncTask.java:305)
       at java.util.concurrent.FutureTask.run(FutureTask.java:237)
       at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:243)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
       at java.lang.Thread.run(Thread.java:761)
```

wcandillon, 2018-07-02:
@sibelius We might be experiencing the same issue, did you find a fix/workaround for it?

andreseiji, 2019-05-20:
```
Caused by: java.lang.IllegalArgumentException: Invalid image data size.
        at com.google.android.gms.vision.Frame$Builder.setImageData(Unkno
        at com.google.firebase.ml.vision.common.FirebaseVisionImage.zza(U
        at com.google.android.gms.internal.firebase_ml.zzpw.zza(Unknown S
        at com.google.firebase.ml.vision.face.FirebaseVisionFaceDetector.Unknown Source)
        at org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInolveTakenPictureAsyncTask.java:109)
        at org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInolveTakenPictureAsyncTask.java:37)
        at android.os.AsyncTask$2.call(AsyncTask.java:295)
        at java.util.concurrent.FutureTask.run(FutureTask.java:237)
```

sibelius, 2019-05-20:
for the fix follow this https://github.com/expo/expo/pull/2772

MateusAndrade, 2019-05-20:
> for the fix follow this [expo/expo#2772](https://github.com/expo/expo/pull/2772)

i can pick this

MateusAndrade, 2019-05-21:
This still happens? I gave a look at ```FaceDetectorAsyncTask``` and there is already a treatment to this:

```java
  @Override
  protected SparseArray<Face> doInBackground(Void... ignored) {
    if (isCancelled() || mDelegate == null || mFaceDetector == null || !mFaceDetector.isOperational()) {
      return null;
    }

    RNFrame frame = RNFrameFactory.buildFrame(mImageData, mWidth, mHeight, mRotation);
    return mFaceDetector.detect(frame);
  }
```

@sibelius do you think wrapping ```doInBackground``` with a try catch is a good addition? Btw, this method is implemented on 11 files with some differences.

What did you do to catch this error?

sibelius, 2019-05-21:
check the expo PR

it is about `BarCodeScannerResult`, not face detection

sibelius, 2019-05-21:
nevermind, the errortrack is on face detector as well

a try catch could help on this one

closed 2019-07-27 by stale[bot]