react-native-camera/react-native-camera #1257: Android version  is not working, giving different Java error messages

donni106, 2018-02-16:
### Which implementation are you using

*RNCamera*

### Behaviour
After updating to master:
My app is running on iOS in a Simulator. No problem.
But i cannot make it work for android.
Tried with many gradle cleans and rebuilds. The linkings should be fine, as far as i can say.
Here are my different error messages, after running react-native run-android or per Run app in android studio.

[image]

What is the problem? I tried with react-native 0.51.0 and updated to 0.53.0. Same behaviour.

I have no idea what to do more. My next step would be a completely new app with latest react-native-camera initially.
It would be nice if it would still work with the current version.

### Environment
- **Node.js version**: v9.5.0
- **React Native version**: 0.53.0
- **React version**: 16.0.0

### react-native-camera
**Version**: "master" 1.0.1

jgfidelis, 2018-02-16:
It is not getting the correct react-native version

closed 2018-02-16 by sibelius

rom1k, 2018-02-18:
@jgfidelis What is the correct react-native version?

donni106, 2018-02-22:
> npm WARN react-native@0.53.0 requires a peer of react@16.2.0 but none is installed. You must install peer dependencies yourself.

Maybe @jgfidelis pointed that. I will give it a try.

donni106, 2018-02-22:
I made it!
There was a bad configuration with **build.gradle** files.
A **react-native@0.20.1** was wrongly selected in the external libraries.
I tried a lot of things, cleaned a lot of times and changed the maven repositories back and forth.
At the end I also set the version in my **app/build.gradle** `compile "com.facebook.react:react-native:0.53.0"` to be sure.
So happy to be finally successful with this after hours of struggling.