react-native-camera/react-native-camera #1258: RNCamera Java won't recognize any QR Code - no matter what is configured

K-Leon, 2018-02-17:
### Which implementation are you using

*RNCamera*

### Steps to reproduce
Just try scanning a QR Code with Java Version of RNCamera. RCTCamera Java recognizes the exact same Barcode. EAN (1D Barcode on a Coca Cola tin) works in RNCamera - QR not.  IOS on the other Hand works on both Platforms.

### Expected behaviour
It should recognize QR Codes

### Actual behaviour
It won't recognize any 1D Barcode

### Environment
- **Node.js version**: Latest
- **React Native version**: Latest
- **React Native platform + platform version**: Android 7

### react-native-camera
**Version**: Current Master

sibelius, 2018-02-17:
this should be related to the aspect ratio choosed, could u try another aspect ratio?

K-Leon, 2018-02-17:
@sibelius 
Thanks for your help! Just checked it with different aspect ratios: (["3:2", "4:3", "11:9", "16:9"] are possible with my camera)

But won't change a thing - except unscannable 1D Barcodes. I think we need some kind of stretch mode like it is applied in RCTCamera if i interpret it's code correctly

jackwilsdon, 2018-02-21:
Experiencing the same issue with `RNCamera`. Tried multiple aspect ratios and nothing worked. Managed to scan a normal 1D barcode but not any QRs.

Skn0tt, 2018-02-23:
For me it's the same, does recognise 1D but no qrs...
What can we do to fix this?

karpie28, 2018-02-25:
With the latest release from master, QRCode scanning on android works for me like a charm, but vertical scanning of EAN-13 barcode not. More here: https://github.com/react-native-community/react-native-camera/issues/1272 .

K-Leon, 2018-02-25:
Which aspect ratios are supported by your device and which got taken?

McBerry, 2018-02-26:
Had the same Issue with recognising QR-Codes. Setting the aspect ratio fixed the problem.

Marian0, 2018-03-16:
Same issue here.

Tested on:

Moto G5s plus
Android 7.1.1

SimonErm, 2018-04-04:
#1422 Try latest master

closed 2018-04-04 by sibelius