react-native-camera/react-native-camera #1259: Updating the Child Elements of RNCamera results in Black Preview Screen

K-Leon, 2018-02-17:
### Which implementation are you using

*RNCamera*

### Steps to reproduce
You need a code construct like this. As soon as you update this.props.children and the inner UI changes the Camera Preview goes black.
```
<RNCamera>
        {this.props.children}
</RNCamera>
```
### Expected behaviour
It should keep working like in RCTCamera

### Actual behaviour
It goes black

### Environment
- **Node.js version**: latest
- **React Native version**: latest
- **React Native platform + platform version**: Android 7

### react-native-camera
**Version**: master

jgfidelis, 2018-02-17:
I think we need a startPreview and stopPreview methods like RCTCamera

K-Leon, 2018-02-17:
I‘m not sure this is the reason - i didn’t used them on rctcamera too. (But maybe it is used internally - i need to look this up)

jgfidelis, 2018-02-17:
Yes, I believe they usem them internally but I have to take a look into the code later.

kosintop, 2018-04-03:
@jgfidelis, may I ask how's the progress on this? I'm having trouble with this issue and would like to know whether it will be possible to render children inside <RNCamera> or not

sibelius, 2018-04-03:
Can somebody debug and try to fix this? I think this only happens when there is a setState

Concert0, 2018-07-13:
It works fine on iOS but not on android(using master branch)

carloscuesta, 2018-08-02:
Having the same problem, @sibelius I'm not using `setState` just a conditional render based on props

markidm, 2018-12-12:
@sibelius I have a conditional render based on state, I'm getting the same issue.

n1ru4l, 2018-12-13:
@markidm Could you please provide a reproduction repository? Or fork this repository in order to implement a reproduction in `examples/basic`?

sibelius, 2018-12-13:
Just do a setState while trying to do a capture

sibelius, 2019-01-08:
fixed here https://github.com/react-native-community/react-native-camera/releases/tag/v1.7.0

closed 2019-01-08 by sibelius