react-native-camera/react-native-camera #1260: RNCamera video not recording

rf1804, 2018-02-19:
Using RNCamera git+https://git@github.com/react-native-community/react-native-camera

I had implemented RCTCamera earlier which worked fine except flash so i'm switching to RNCamera but i'm not able to start video recording and there is not example on Video recording
can anyone who is able to record video using RNCamera can post it so it can be sued as example for implementation.

VrajSolanki, 2018-02-21:
`<Camera
                ref={(cam) => {
                    this.cameraComponent = cam;
                }}
                captureMode={this.props.mode == 'image' ? Camera.constants.CaptureMode.still : Camera.constants.CaptureMode.video}
                style={styles.preview}
                type={"back"}
                captureQuality={"high"}
                captureAudio={true}
                fixOrientation={true}
                aspect={Camera.constants.Aspect.fill}>
                   
                    <View style={ this.props.mode == "video" && this.state.isRecording ? styles.footerContainerWithoutControls : styles.footerContainer}>
                        {
                            this.state.isRecording ? 
                            <TouchableOpacity style={styles.videoSvg} onPress={() => this.endVideoCapture()}><RecordStop /></TouchableOpacity>
                            :
                                <TouchableOpacity style={styles.videoSvg} onPress={() => this.takePicture()}><RecordStart /></TouchableOpacity>

                        }
                    </View>
                </Camera>`

`
        endVideoCapture = () => {
        this.setState({isRecording: false})
        this.cameraComponent.stopCapture()
    }
      takePicture = async function() {
    if (this.camera) {
      this.cameraComponent.capture({metadata: options})
            .then((data) => {
                Image.getSize(data.path, (width, height) => {
                    const capturedObject = {
                        url: data.path,
                        name: `RandomName`,
                    }
                    console.log(capturedObject);
                }, console.log);
                })
            .catch(err => console.error(err));
}
`

closed 2018-03-02 by jgfidelis

Deepak-merahkee, 2019-09-25:
RNCamera works fine please check here [https://stackoverflow.com/a/58099743/8746415](https://stackoverflow.com/a/58099743/8746415) , it gives clear picture how to user RNCamera for video recording. comment there for more details