react-native-camera/react-native-camera #1262: Pluggable Extension System

ma-pe, 2018-02-19:
Hi all!

@sibelius and I were thinking about a system of extensions for react-native-camera.

It would be nice if users could choose extensions as post-processing (`Google Vision`, `OpenCV`, ...) or any other extension (ideally community written). These extensions would be optional resulting in a minimal setup with minimal footprint for users that just need the camera without extensions.

Is there anyone that could give some insights about how to structure a system like that for iOS/ObjectiveC and Android/Java? Or even better: anyone with experience who'd like to help?

jgfidelis, 2018-02-19:
I think we could leave the basic `react-native-camera` without any extension as a package. Using it would be as easy as adding the package and linking it.

Then, create another repo for camera with exentions, like Face Detection, that already adds some bit difficult steps to the installation process.

For iOS/ObjC we could make many targets and select the files to compile to each target. The user would then add the correct targets to its app's xcode project, but I am still a junior developer so anyone with more experience may have a better idea.

n1ru4l, 2018-12-10:
## iOS

On Objective-C we can do something like:

```objc
#if __has_include(<react-native-camera/ExtensionXY.h>)
```

**Default Setup:**
With https://github.com/react-native-community/react-native-camera/issues/1980 we could make enabling the extensions easier.

**CocoaPods Setup:**

Here we can use SubPods to enable specific features (I already did something similar for react-native video https://github.com/react-native-community/react-native-video/blob/b2aa8c4e383047475fe6ad794fdf40f058bdec9b/react-native-video.podspec#L23)

## Android

I currently have no Idea how to conditionally add extensions **Help Wanted**

____

Personally I would also prefer if all features that are not basic should not be enabled/implemented by default