react-native-camera/react-native-camera #1263: Image has weird properties, causing flipped images

benjdlambert, 2018-02-19:
### Which implementation are you using
*RNCamera*

### Steps to reproduce
1. Take picture to base64

### Expected behaviour
Image is displayed properly.

### Actual behaviour
- So this is an interesting one that I can't really explain.
- The image seems to be rotated 90 degrees when it's used in places.
- I can provide an example here: (I just pasted this link in here: `s3-eu-west-1.amazonaws.com/picture.dev.justfootball.io/8b7c50a9-5eed-4874-a619-c96795ebebb2/8212e579-
0c62-430b-86dc-42360be8928a`)

Notice the difference when it's loaded in Chrome, when something else goes to fetch it.

[image]

### Environment
- **Node.js version**: 8
- **React Native version**: 0.52.0
- **React Native platform + platform version**: iOS 11.2.1 not tested on android
### react-native-camera
**Version**: "master" but happens on ^1.0.1 also

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

jgfidelis, 2018-02-19:
Can you please test on Android as well?

johnbodenhamer, 2018-02-20:
I am having this problem on a Pixel 2

nkov, 2018-02-21:
We are seeing this issue on Android and iOS, and we have checked the EXIF properties which show a field `Orientation: Rotate 90 CW`

Michaelvilleneuve, 2018-02-23:
can you try with forceUpOrientation using 
`"react-native-camera": "git+https://git@github.com/Snapp-FidMe/react-native-camera",`
see https://github.com/react-native-community/react-native-camera/pull/1278 for more info

Fconstant, 2018-02-27:
Fixed by #1290

closed 2018-02-27 by Fconstant

arapocket, 2018-03-07:
Is anyone still getting landscape photos even when using forceUpOrientation?

Testing on a simulator it worked properly. Testing on an Iphone X it did not. Has anyone had trouble with any other devices?

aditya-rumbleon, 2018-05-18:
I still see the issue with landscape photo after using forceUpOrientation. Testing on iPhone 6S

n1mda, 2019-01-17:
I'm still seeing this issue, even with `forceUpOrientation: true` setting

saifali96, 2019-10-01:
We are facing the same issue, but only on Samsung Galaxy S10+ device.. anyone with a clue why?