react-native-camera/react-native-camera #1267: Camera preview is distorted

sandropoluan, 2018-02-20:
### Expected behaviour
Camera preview should not distorted

### Actual behaviour
Camera preview distorted

### Environment
- **Node.js version**: v8.9.3
- **React Native version**: "0.53.0"
- **React Native platform + platform version**: Tested on Android 5.0, Android 7.1 etc

### react-native-camera
**Version**: 1.0.1 (NPM)

### My Code
```
          return (<View style={{flex:1}}>
                       <RNCamera ref={ref => {
                          this.camera = ref;
                        }}
                        style={{flex: 1,justifyContent: 'flex-end',alignItems: 'center'}}
                        type={RNCamera.Constants.Type.back}
                        permissionDialogTitle={'Permission to use camera'}
                        permissionDialogMessage={'We need your permission to use your camera phone'}                    
                        
                        />
                     </View>);
```

Object is tall and slim
[image]

When I rotate it become short and wide.
[image]

alizahid, 2018-02-21:
I have a square-sized camera preview, and I have this issue, too. I had a hacky fix, which involved calculating the layout height and width and setting on the camera preview, and then setting the ratio to `3:4`, and it worked. On some devices, at least

Today, I upgraded to `1.0.1` and it crashes when trying to render camera preview. A similar stack trace here https://github.com/google/cameraview/issues/206

Using `RNCamera`

jgfidelis, 2018-02-21:
Some of this errors are likely due to CameraView, but we use Expo's version, so I do not know the best place to open issues about it, if it is in Expo's repo or Google's repo.

jgfidelis, 2018-02-21:
@alizahid did you use `getSupportedRatiosAsync` before setting the ratio that crashed the device?

alizahid, 2018-02-21:
It was indeed an unsupported ratio that caused the crash.

waitopiggu, 2018-02-22:
On my Android development phone, setting the aspect to '16:9' removed the distortion.

Without some way of getting the dimensions from the camera itself, there's no way to work out the correct aspect ratio to use.

Allowing the user to choose from a list of supported aspect ratios would work, but not a very good solution from a UX perspective.

alizahid, 2018-02-25:
@waitopiggu `3:4` worked for me. But it's not available on most devices.

Sangrene, 2018-03-28:
Hi guys, just a quick workaround ONLY if you render it on all the screen with position absolute: 
In your render camera function write this at the beginning
```
const {height, width} = Dimensions.get('window');
const newWidth = height*(3/4);
const widthOffset = -((newWidth-width)/2);
```

Then in your RNCamera component, add the follow style
`{left: widthOffset, right:widthOffset}`

Just for you to know, my base RNCamera style is
```
preview: {
    position: "absolute",
    top: 0,
    bottom: 0,
    right: 0,
    left: 0,
    zIndex: 2
  }
```

Hope it helps you :)

jgfidelis, 2018-04-04:
please try on master branch

closed 2018-04-04 by sibelius

gigby, 2018-05-07:
It is not a bug. Please, check here https://github.com/react-native-community/react-native-camera/issues/1280