react-native-camera/react-native-camera #1268: Xcode expected a type

johnbodenhamer, 2018-02-20:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Which implementation are you using
version 1.0.1
*RNCamera* 

### Steps to reproduce
followed all installation steps up to remove FaceDetector
when attempting to build I get an error "Expected a type"
``` 
+ (AVCaptureDevice *)deviceWithMediaType:(AVMediaType)mediaType preferringPosition:(AVCaptureDevicePosition)position // <-- Right here
{
    NSArray *devices = [AVCaptureDevice devicesWithMediaType:mediaType];
    AVCaptureDevice *captureDevice = [devices firstObject];
    
    for (AVCaptureDevice *device in devices) {
        if ([device position] == position) {
            captureDevice = device;
            break;
        }
    }
    
    return captureDevice;
}
```
### Expected behaviour
project should compile

### Environment
- **Node.js version**:5.6.0
- **React Native version**:0.52.0
- **React Native platform + platform version**: iOS 9.0

### react-native-camera
**Version**: 1.0.1

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

sibelius, 2018-02-20:
this old happens on old xcodes

johnbodenhamer, 2018-02-20:
Yes, that worked. Thank you

closed 2018-02-20 by johnbodenhamer