react-native-camera/react-native-camera #1271: Cannot specify desired resolution with RNCamera

formigone, 2018-02-21:
I can't seem to find the docs that specify how to configure the desired resolution of the output picture taken with RNCamera. Is this even possible out of the box? I'm only able to take 3000x4000 plus images on my Pixel. Ideas?

jgfidelis, 2018-02-21:
You cannot. You have to take the picture and use another lib to resize it to the size you want.

Someone could send a PR of this feature as a TakePicture Option

formigone, 2018-02-21:
Thanks for your reply. I'd love to take a stab at implementing this feature - at least for Android at first, as that's what I'm experienced with.

sibelius, 2018-02-21:
@formigone RNCamera is based on Expo camera module: https://github.com/expo/expo

formigone, 2018-02-21:
For those interested, I came across this not-quite-ideal, but "native" solution (requires to additional dependencies):

 1. Capture photo using `RNCamera`
 2. Use `react-native.ImageEditor.cropImage` to "fake crop" image (from position `0x0` to `WIDTHxHEIGHT`, using the `displaySize` option, which resizes the output image)

---
     this.camera.takePictureAsync(options)
         .then(({ uri, width, height }) => {
            ImageEditor.cropImage(uri, {
               offset: { x: 0, y: 0 },
               size: { width, height },
               displaySize: { width: TARGET_WIDTH, height: TARGET_HEIGHT},
            }, (res) => {
               // res == 'file:///data/.../img.jpg'
            }, (err) => {
               // ='(
            }
         });

closed 2018-02-21 by formigone

sibelius, 2018-02-21:
@formigone can u add this do `docs`?

maybe we can have a list of `recipes` or snippets to solve common cases

formigone, 2018-02-21:
You bet. I'll have a PR ready later today.

cristianoccazinsp, 2019-05-11:
Is there any technical reason to not expose functionality to set the camera resolution at capture time? Resizing the image pretty much doubles the capture time.