react-native-camera/react-native-camera #1272: EAN-13 barcode doesn't scan vertically

karpie28, 2018-02-21:
### Steps to reproduce
1. Try to scan barcode EAN13 format
2. Present barcode vertically (does not scan)
3. Present barcode horizontally (Scans quickly)

### Expected behaviour
Should scan both horizontally and vertically

### Actual behaviour
Present barcode horizontally (does not scan)
Present barcode vertically (Scans quickly)

When book is vertically:

[image]

When book is horizontally:

[image]

### Environment
- **Node.js version**: 8.4.0
- **React Native version**: 0.50.4
- **React Native platform + platform version**:  Android 6.0.1

### react-native-camera
**Version**: master

CodeAddictSebastianGolinski, 2018-02-21:
Same problem here 😕

OzoTek, 2018-02-22:
Same, since I upgraded from RCTCamera to RNCamera last version, was working fine before

K-Leon, 2018-02-25:
Probably related to aspect ratio like discussed in #1258

JorgeCeja, 2018-03-01:
What aspect ratio are you using? @karpie28

csumrell, 2018-03-02:
I am getting this same error on Android only, and only when my phone is positioned in Vertical/Portrait mode. It just stares at the barcode without doing anything, however if i stop the camera, turn the phone sideways (Landscape) and scan it, it processes it very quickly.

movie4, 2018-03-09:
Get seems issues on RNCamera 1.0.1 - EAN13 — barcode

SimonAmpleman, 2018-03-09:
Also have the same problem. Application is in portrait mode. I've tried every aspect ratio possible. If I turn the phone in landscape, it works fine.

Also, when I save a picture, it appears in landscape mode (it's rotated 90 degrees).

Thank you for your help!

SimonAmpleman, 2018-03-09:
I'll contribute a 50$ USD if this can be fixed : Camera in portrait mode. Taking picture in portrait mode. And detecting UPC in both portrait or orientation at the same time.

Thanks !

ChangeSomeCode, 2018-03-11:
same problem

joselazarosiqueira, 2018-03-12:
Same problem!

uikolas, 2018-03-23:
Same on android

krystianpach, 2018-03-26:
Same issue on all Android Devices, there is no problem with iOS

DanielLeberle, 2018-03-28:
I have the same problem, does anyone have not this problem?

E-P1983, 2018-03-29:
Same problem!

Hoolii, 2018-03-29:
Same here, since I migrated from RCTCamera to RNCamera.

DanielLeberle, 2018-03-30:
Does anyone have a solution to make scanning possible in portrait mode but not in landscape mode? this would help me a lot.

I had a look into the java code and tried to rotate the image but without success.

jpudysz, 2018-03-30:
@DanielLeberle yes. Downgrade to `RCTCamera`

DanielLeberle, 2018-03-30:
well I guess there is no other option. Although RCTCamera has not the same performance as RNCamera in my opinion. On iOS RNCamera is a lot better, thats why we switched.

jpudysz, 2018-03-30:
I agree, but we end up using 2 cameras. One for barcode scanner and second for photos. Still it's better to have app than crashes :)

I guess we need to wait ;)

MWalid, 2018-04-01:
same here.... +1

jgfidelis, 2018-04-02:
Please try on master branch

Ilario17, 2018-04-03:
Now it works in portrait but on my Wiko Rainbow the performances are terrible

platonish, 2018-06-30:
Still doesn't work. Tested on Android Nougat, react-native-camera v1.1.4

prateekgoel, 2018-12-12:
@sibelius This is still the issue for me. I have tried it with the latest version of react-native-camera and with RNCamera and RCTCamera also. Not working in any case.

Eriwyr, 2019-02-06:
Same problem for me, anyone has found a way to fix this problem ?

JohnTree, 2019-03-27:
Same problem for me, anyone has found a way to fix this problem ?

sibelius, 2019-04-18:
RCTCamera is removed https://github.com/react-native-community/react-native-camera/issues/1890

Migrate to latest version of RNCamera

wolkingdead, 2019-08-27:
same problem how to fix it

luizaprata, 2020-06-24:
same problem how to fix it

closed 2020-08-30 by stale[bot]