react-native-camera/react-native-camera #1279: TypeError: this.camera.takePictureAsync is not a function

robisaks, 2018-02-22:
### Which implementation are you using

*RNCamera* 

### Steps to reproduce
1. install 1.0.1 or Master
2. Call takePictureAsync
3.

### Expected behaviour
Picture taken

### Actual behaviour
TypeError: this.camera.takePictureAsync is not a function

### Environment
- **Node.js version**: v8.9.1
- **React Native version**:  0.51.0-rc.2
- **React Native platform + platform version**: Android 8.1.0

### react-native-camera
**Version**: 1.0.1 or Master

### Info
Kind of an odd issue. I looked at the RNCamera.js file in node_modules/react-native-camera and it has the function. "this.camera" appears to have some of the methods but not that one. Another oddity is that I had to set the "constants" in RNCamera.constants to be lower case, other wise it would say it was undefined. I've tried deleting the node_mocules/react-native-camera and re-installing. Not sure what is going on.

```
takePicture = async function() {
    console.log(this.camera);
    if (this.camera) {
      try {
        const options = { quality: 0.5, base64: true };
        const data = await this.camera.takePictureAsync(options)

        console.log(data.uri);
        // this.setCameraCaptureVisible(false);
      } catch (e) {
       // This logs the error
        console.log(e)
      }

    }
  };
```
```
<RNCamera
  ref={(cam) => {
    this.camera = cam;
  }}
  type={RNCamera.constants.Type.back}
  style={styles.preview}
  aspect={RNCamera.constants.Aspect.fill}>
</RNCamera>
```

 ``` 
<TouchableOpacity onPress={this.takePicture.bind(this)}>
      <Image source={require('./img/ic_camera_alt_2x.png')} style={styles.actionButtons}/>
  </TouchableOpacity>
```

sibelius, 2018-02-22:
The link is wrong

robisaks, 2018-02-22:
In package.json? I've tried:
`"react-native-camera": "git://github.com/lwansbrough/react-native-camera.git#master",`
and 
`"react-native-camera": "^1.0.1",`

robisaks, 2018-02-22:
After more digging, it looks like _Camera.js_ does not contain _takePictureAsync_, that is in RNCamera.js. But it looks like my project is using _Camera.js_ instead of the Camera class inside of _RNCamera.js_. Why would that be?

robisaks, 2018-02-22:
Solved! Thank you for your help! I appreciate the super quick response, it put me on the right path.
Switched 
`import RNCamera from 'react-native-camera';` 
to 
`import { RNCamera } from 'react-native-camera';`

closed 2018-02-22 by robisaks