react-native-camera/react-native-camera #1280: RNCamera ratio question

Jinran, 2018-02-23:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Which implementation are you using

RNCamera

### Expected behaviour
I need a camera view with changeless width & height,
eg: width: 200,height:100

### Actual behaviour
this is iphone7 device
[image]
this is huawei P9 device
[image]

why my camera streched like this and how can I set ratio to solve this problem in android with changeless width & height

### Environment
- **Node.js version**:
- **React Native version**:
- **React Native platform + platform version**: iOS 11.0+, Android 7.0, etc

### react-native-camera
**Version**: npm version or "master"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

Fconstant, 2018-02-26:
Android has a unique prop `ratio` which default value is `4:3`.
Try to get all supported ratios before, and then setting it to your desire.

Example:
```javascript
import React from 'react';
import { Platform } from 'react-native';
import { RNCamera } from 'react-native-camera';

const DESIRED_RATIO = "16:9";

class MyCameraComponent extends Component {
    state = {}

    prepareRatio = async () => {
        if (Platform.OS === 'android' && this.cam) {
             const ratios = await this.cam.getSupportedRatiosAsync();

             // See if the current device has your desired ratio, otherwise get the maximum supported one
             // Usually the last element of "ratios" is the maximum supported ratio
             const ratio = ratios.find((ratio) => ratio === DESIRED_RATIO) || ratios[ratios.length - 1];
             
             this.setState({ ratio });
        }
    }

    render() {
        return (
            <RNCamera
                ref={(cam) => this.cam = cam}
                onCameraReady={this.prepareRatio} // You can only get the supported ratios when the camera is mounted
                ratio={this.state.ratio}
            />
        );
}
```

Fconstant, 2018-02-27:
@Jinran Please, close this issue if this is not a problem anymore. Thanks :)

closed 2018-02-28 by sibelius