react-native-camera/react-native-camera #1281: Is there a way to turn off capture sound?

bvv218, 2018-02-23:


sibelius, 2018-02-23:
there is a `mute` option on RNCamera

bvv218, 2018-02-23:
@sibelius It doesn't seem to work when you take pictures. Doesn't it mute sound of videos only?

ajaxmac, 2018-02-23:
Not on iOS, this is enforced by the operating system. I saw some notes on turning off the shutter sound for Android somewhere. Apparently it's law in Japan that phone cameras must make a sound so that people can't surreptitiously take photos.

jgfidelis, 2018-02-23:
ios has the prop `captureAudio` as `false` as default. If you want to capture audio, pass `true`
 read the doc https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md

closed 2018-02-23 by sibelius

SBright, 2020-12-30:
Is it possible to disable capture sound on RNCamera in iOS?