react-native-camera/react-native-camera #1282: Pinch Zoom

sachaBlitz, 2018-02-23:
### Which implementation are you using

*RNCamera*

### Expected behaviour
I would like to have a "pinch zoom", is it possible to have it with this version ?

### react-native-camera
1.0.1

Thanks you,

Kind regards,

jgfidelis, 2018-02-23:
Not in this version, but would be a nice feature to have. Send a pr if you know how to implement this.

klaasman, 2018-02-27:
Am I right there actually was some sort of pinch zoom implementation in the `RCTCamera`?

n1ru4l, 2018-05-28:
@jgfidelis I need this feature for our app, so I will do some research.

There is already a `zoom` property on the RNCamera component. It might be better to do the zoom stuff on the native side to make it more fluent. I will try to do a iOS implementation asap.

sibelius, 2018-05-28:
can't we implement this on user land?

can we make this work using pan responders?

check this https://github.com/ascoders/react-native-image-zoom

n1ru4l, 2018-05-28:
@sibelius I think we should decide this after checking if the transitions are fluid without native code 😅

n1ru4l, 2018-05-29:
Using a `PanResponder` for updating the zoom property works pretty good. We decided to use a different gesture for zooming (swiping up while the record button is pressed).

We are using this utility component:

```js
import React, { Component } from "react";
import { View, PanResponder, Dimensions } from "react-native";

class ZoomView extends Component {
  constructor(props) {
    super(props)
    this._panResponder = PanResponder.create({
      onPanResponderMove: (e, { dy }) => {
        const { height: windowHeight } = Dimensions.get(`window`)
        return this.props.onZoomProgress(
          Math.min(Math.max(dy * -1 / windowHeight, 0), 0.5),
        )
      },
      onMoveShouldSetPanResponder: (ev, { dx }) => {
        return dx !== 0
      },
      onPanResponderGrant: () => {
        return this.props.onZoomStart()
      },
      onPanResponderRelease: () => {
        return this.props.onZoomEnd()
      },
    })
  }
  render() {
    return (
      <View
        style={{ flex: 1, width: `100%` }}
        {...this._panResponder.panHandlers}
      >
        {this.props.children}
      </View>
    )
  }
}

```

Sine different people have different gesture requirements I would agree with @sibelius to keep this in the user land. Maybe someone could cook up a recipe document.

JB-CHAUVIN, 2018-08-09:
Anything new ?

n1ru4l, 2018-09-24:
@JB-CHAUVIN This should be implemented in user land. Please feel free to use/adjust my example.

I am going to close this issue.

closed 2018-09-24 by n1ru4l

sibelius, 2018-10-02:
@n1ru4l maybe we should add your code do docs/

n1ru4l, 2018-10-02:
Sure I will make a pull request with our updated version this month! #hacktober

msageryd, 2018-10-25:
Please correct me if I'm wrong.. If this is implemented in JS, is it really possible to utilize the optical 2x zoom in iPhoneX and iPhone8, and other models? 

A JS zoom is merely a crop, isn't it?

jasan-s, 2019-01-17:
@n1ru4l how exactly does the utility function you shared is integrated with RNCamera?

n1ru4l, 2019-01-17:
You pass a `onZoomProgress` function to the component and set the value as the state which is then passed to RNCamera.

```jsx
import React, { Component } from "react";
import { RNCamera } from 'react-native-camera'
import { ZoomView } from "./zoom-view"

class CameraScene extends Component {
  state = {
    zoom: 1
  }
  render() {
    return <ZoomView onZoomStart={() =>{}} onZoomEnd={() => {}} onZoomProgress={zoom => this.setState({ zoom })}><RNCamera zoom={this.state.zoom}>{this.props.children}</RNCamera></ZoomView>
  }
}
```

chadhobson, 2019-01-22:
@n1ru4l I'm guessing you aren't using this in a modal, eh? Anyone know how to get PanResponder to work in a modal?

cristianoccazinsp, 2019-05-14:
Adding to @msageryd , I have used the following code to handle pinch events. Not completely tested, but adding this to RNCamera children and implementing some of the listeners should be straight forward.

```
import React from 'react';
import {View} from 'react-native';

import {
  PinchGestureHandler,
  State,
} from 'react-native-gesture-handler';

export default class ZoomView extends React.Component {

  onGesturePinch = ({ nativeEvent }) => {
    this.props.onPinchProgress(nativeEvent.scale);
  }

  onPinchHandlerStateChange = event => {

    if (event.nativeEvent.state === State.END){
      this.props.onPinchEnd();
    }
    else if (event.nativeEvent.oldState === State.BEGAN && event.nativeEvent.state === State.ACTIVE){
      this.props.onPinchStart();
    }
  };

  render() {
    return (
      <PinchGestureHandler
        onGestureEvent={this.onGesturePinch}
        onHandlerStateChange={this.onPinchHandlerStateChange}
      >
        <View style={this.props.style}>
          {this.props.children}
        </View>
      </PinchGestureHandler>
    );
  }
}

ZoomView.defaultProps = {
  onPinchProgress: (p) => {},
  onPinchStart: () => {},
  onPinchEnd: () => {}
}
```

cristianoccazinsp, 2019-05-14:
Not remotely close as doing it on the native side though. Calling set state on the RN camera component on each zoom change is quite slow.

jqn, 2019-10-03:
I second @cristianoccazinsp the js implementation feels very clunky, also it make the app crash if there are too many state changes while zooming in/out

cristianoccazinsp, 2019-10-03:
Haven't tried it, but using setNativeProps might help

fabriziobertoglio1987, 2019-10-14:
> I second @cristianoccazinsp, the Javascript implementation feels very clunky, also the app crashes if there are too many state changes while zooming in/out

@cristianoccazinsp so there is no possibility to change the zoom natively without writing native code? That would be the solution right?

No guide or medium article on how to do that? 
Thanks

fabriziobertoglio1987, 2019-10-15:
> I second @cristianoccazinsp, the Javascript implementation feels very clunky, also the app crashes if there are too many state changes while zooming in/out

@jqn @cristianoccazinsp 

I did not test this functionalities, but I downloaded the Synced app which implements the following solution

https://medium.com/synced-to-tech-blog/implementing-pinch-to-zoom-in-react-native-8a291ca29ac1

I tested their camera and the zoom in/out effect works perfectly on android

https://play.google.com/store/apps/details?id=to.synced.synced&hl=en

fabriziobertoglio1987, 2019-10-15:
> I second @cristianoccazinsp, the Javascript implementation feels very clunky, also the app crashes if there are too many state changes while zooming in/out

@jqn @cristianoccazinsp I was [reading this post](https://medium.com/synced-to-tech-blog/implementing-pinch-to-zoom-in-react-native-8a291ca29ac1) and I [downloaded their app](https://play.google.com/store/apps/details?id=to.synced.synced&hl=en) 

Unluckily the app source code is not available, the camera zoom works perfectly, much better then react-native camera.

I don't think that we need to use react-native-camera to take a picture or a video. 
We can just create an intent so the user opens the camera application, records the video and then return us the uri of the video.

This implementation would be much easier to build then react-native-camera

cristianoccazinsp, 2019-10-15:
That might work for some applications, but if you need to take pictures from within the app, rapidly, without switching apps, that would not be an option.

I've been using just the zoom prop and making sure that the component rendering is performant enough and I haven't had much issues with the zoom anymore. I'm also using react-native-gesture-handler to handle the pinch events efficiently.

fabriziobertoglio1987, 2019-10-15:
@cristianoccazinsp 

Maybe not using react-native-camera and instead [just writing an intent to open the default camera application](https://stackoverflow.com/a/2729382/7295772)

[The following app](https://synced.to/) uses this solution, all the camera functionalities work perfectly.

[image]

Could you re-open the issue as it seems to not be solved? 
Developers may need to review all the open issues to estimate the project deadlines.

Thanks

fabriziobertoglio1987, 2019-10-17:
@cristianoccazinsp 

this is my existing code. Can you help me completing this functionality? Thanks

```
import React, { Component } from 'react';
import { View } from 'react-native';
import { PinchGestureHandler, State } from 'react-native-gesture-handler';

export default class ZoomView extends Component {
  onGesturePinch = ({ nativeEvent }) => {
    this.props.onPinchProgress(nativeEvent.scale)
  }

  onPinchHandlerStateChange = event => {
    const pinch_start = event.nativeEvent.state === State.END
    const pinch_begin = event.nativeEvent.oldState === State.BEGAN
    const pinch_active = event.nativeEvent.state === State.ACTIVE
    if (pinch_start) {
      this.props.onPinchEnd()
    }
    else if (pinch_begin && pinch_active) {
      this.props.onPinchStart()
    }
  }

  render() {
    const { style } = this.props
    return (
      <PinchGestureHandler
        onGestureEvent={this.onGesturePinch}
        onHandlerStateChange={this.onPinchHandlerStateChange}>
        { this.props.children }
      </PinchGestureHandler>
    )
  }
}
```

and the `Camera` Component

```
import React, { Component } from 'react'
import { RNCamera } from 'react-native-camera';
iimport ZoomView from './ZoomView';

export default class Recorder extends Component {
  _render() {
    return (
      <ZoomView
        onPinchStart={() => {}}
        onPinchEnd={() => {}}
        onPinchProgress={zoom => this.setState({ zoom })}>
        <RNCamera
          ref={ref => {
            this.camera = ref;
          }}
          zoom={this.state.zoom} />
      </ZoomView>
    )
  }
}
```

cristianoccazinsp, 2019-10-17:
I have something like this, you need to take the pinch travel distance and map it to zoom value accordingly. There are probably way better ways to do this, but this is what I'm using so far. ZOOM_F I think could be a bit smaller on Android as well to give a smoother experience, but since android devices are usually slower, and my component is not very efficient, I decided to make it that way.

You can also try using setNativeProps instead of setState if you don't need to display the zoom value on the UI, or you can throttle the setState or something... I honestly didn't need it for now, but setNativeProps should make it way more efficient. Something like `this.camera.setNativeProps({zoom: zoom})`

```
const MAX_ZOOM = 8; // iOS only
const ZOOM_F = IS_IOS ? 0.01 : 0.1;

<RNCamera zoom={this.state.zoom} maxZoom={MAX_ZOOM} ...others><ZoomView ...events style></ZoomView></RNCamera>

this.state = {
    zoom: 0,
    ... others
}

onPinchStart = () => {
    this._prevPinch = 1;
}

onPinchEnd = () => {
    this._prevPinch = 1;
 }}
  

onPinchProgress = (p) => {
    let p2 = p - this._prevPinch;

    if(p2 > 0 && p2 > ZOOM_F){
      this._prevPinch = p;
      this.setState({zoom: Math.min(this.state.zoom + ZOOM_F, 1)})
    }
    else if (p2 < 0 && p2 < -ZOOM_F){
      this._prevPinch = p;
      this.setState({zoom: Math.max(this.state.zoom - ZOOM_F, 0)})
    }
}
```

And the ZoomView.js is pretty much the same as yours, but here it is.

```
import React from 'react';
import {View} from 'react-native';

import {
  PinchGestureHandler,
  State,
} from 'react-native-gesture-handler';

export default class ZoomView extends React.Component {

  onGesturePinch = ({ nativeEvent }) => {
    this.props.onPinchProgress(nativeEvent.scale);
  }

  onPinchHandlerStateChange = event => {

    if (event.nativeEvent.state === State.END){
      this.props.onPinchEnd();
    }
    else if (event.nativeEvent.oldState === State.BEGAN && event.nativeEvent.state === State.ACTIVE){
      this.props.onPinchStart();
    }
  };

  render() {
    return (
      <PinchGestureHandler
        onGestureEvent={this.onGesturePinch}
        onHandlerStateChange={this.onPinchHandlerStateChange}
      >
        <View style={this.props.style}>
          {this.props.children}
        </View>
      </PinchGestureHandler>
    );
  }
}

ZoomView.defaultProps = {
  onPinchProgress: (p) => {},
  onPinchStart: () => {},
  onPinchEnd: () => {}
}
```

fabriziobertoglio1987, 2019-10-18:
the solution was provided by @cristianoccazinsp with the following [answer](https://github.com/react-native-community/react-native-camera/issues/1282#issuecomment-543188018)

Initial testing without using `this.camera.setNativeProps({zoom: zoom})` have smooth zoom effect on even old android and iphone 5 devices (as displayed in the below gif)

Thanks a lot for the support!

[image]

cristianoccazinsp, 2019-10-18:
I see you decreased the ZOOM_F even more than me, did it end up with a smoother zoom transition? Have you tried using setNativeProps to see if it is any faster? When I tried 0.001 it was definitely way too many setState calls for me.

fabriziobertoglio1987, 2019-10-19:
My last setting was 

```javascript
const ZOOM_F = Platform.OS === 'ios' ? 0.0005 : 0.08;
```

It works on iPhone 5, which has a very small screen and was running smooth, but I will try with the setNativeProps now and attach a gif of the iPhone functionality.

I only have an iPhone 5 to test the functionality and it is not possible to run the Camera on the xcode Simulator, so I do not know if it will work on the iPhone X bigger screen, as the motion may span a larger distance.

Thanks a lot for the help

fabriziobertoglio1987, 2019-10-19:
I was reading about `setNativeProps` in the [`react-native-documentation`](https://facebook.github.io/react-native/docs/direct-manipulation.html). 

As my understanding `setNativeProps` is implemented in the [native ios and android component of `TouchableOpacity`](https://facebook.github.io/react-native/docs/direct-manipulation.html#setnativeprops-with-touchableopacity) to change the Opacity of the child element. 

On the `javascript` side there is no effective logic, but the code is implemented on the native level.

I notice that the [`RNCamera` Component passes `...nativeProps`](https://github.com/react-native-community/react-native-camera/blob/master/src/RNCamera.js#L773)

```javascript
  _convertNativeProps({ children, ...props }: PropsType) {
    const newProps = mapValues(props, this._convertProp);

    if (props.onBarCodeRead) {
      newProps.barCodeScannerEnabled = true;
    }

    if (props.onGoogleVisionBarcodesDetected) {
      newProps.googleVisionBarcodeDetectorEnabled = true;
    }

    if (props.onFacesDetected) {
      newProps.faceDetectorEnabled = true;
    }

    if (props.onTextRecognized) {
      newProps.textRecognizerEnabled = true;
    }

    if (Platform.OS === 'ios') {
      delete newProps.googleVisionBarcodeMode;
      delete newProps.ratio;
    }

    return newProps;
  }
```

The props are passed then to the [`RNCamera` native component](https://github.com/react-native-community/react-native-camera/blob/master/src/RNCamera.js#L835).

`this.camera.setNativeProps()` is not a `function` so I believe the key is passing it as the correct `prop` to `RNCamera` so that is converted as nativeProp.

The props passed to the [`RNCamera` native component](https://github.com/react-native-community/react-native-camera/blob/master/src/RNCamera.js#L835) are then used from the [`RNCamera.m` ios component](https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L51)

The ios class has an [`updateZoom` method for re-rendering and changing the zoom](https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L516)

cristianoccazinsp, 2019-10-19:
Interesting. I thought `setNativeProps` was available to any native RN component as a shortcut to using setState. Is it because RNCamera actually wraps the native component with another component, so the ref used is not really the native component ref? If that's the case, then the RNCamera wrapper should just forward any setNativeProps to it's native component.

fabriziobertoglio1987, 2019-10-21:
>so the ref used is not really the native component ref?

I don't know, `RNCamera.js` is long file, but I believe the key to understand it this is here:

```javascript
const RNCamera = requireNativeComponent('RNCamera', Camera, {
  nativeOnly: {
    accessibilityComponentType: true,
    accessibilityLabel: true,
    accessibilityLiveRegion: true,
    barCodeScannerEnabled: true,
    googleVisionBarcodeDetectorEnabled: true,
    faceDetectorEnabled: true,
    textRecognizerEnabled: true,
    importantForAccessibility: true,
    onBarCodeRead: true,
    onGoogleVisionBarcodesDetected: true,
    onCameraReady: true,
    onPictureSaved: true,
    onFaceDetected: true,
    onLayout: true,
    onMountError: true,
    onSubjectAreaChanged: true,
    renderToHardwareTextureAndroid: true,
    testID: true,
  },
});
```
The [`Camera` component is here defined](https://github.com/react-native-community/react-native-camera/blob/master/src/RNCamera.js#L339).

as `this.camera.setNativeProps()` does not work, I may try using a [`forward-ref`](https://reactjs.org/docs/forwarding-refs.html) if that is how you use `setNativeProps`.

I wonder if it is possible to `setNativeProps` on the [`Camera` ios class](https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L51)

Ashoat, 2019-11-02:
I recently worked on getting pinch to zoom to work well on older Android phones (4.4). Besides throttling the `zoom` prop updates, performance can be improved by reducing bridge chatter. I'm using `react-native-reanimated` to filter the throttled events out before they reach the JS side, instead of filtering on the JS side. My approach doesn't decrease the number of React renders, but in my testing it significantly improved performance on older phones. I think this is probably because the level that `react-native-gesture-handler` throttles JS-callback-as-events is probably optimized for newer phones.

You can see my approach [here](https://github.com/Ashoat/squadcal/blob/master/native/media/camera-modal.react.js)

fabriziobertoglio1987, 2019-11-03:
Currently I am testing my functionality on a newly purchased Iphone 6.
I had to change the `ZOOM_F` constant from `0.001` to `0.0008`. Previously I tested the settings with an iPhone 5 and it was acceptable (but already laggy).

```javascript
const ZOOM_F = Platform.OS === 'ios' ? 0.0008 : 0.08;
```

I think this functionality could be grately improved by detecting if the user stops `zooming in` and starts `zooming out`. 

The zoom-in effect should stop once the user start pinching in the opposite direction. 

It would improve greately the functionality. 

I am also worried of how the `ZOOM_F` constant renders on different phones.
We are all using different settings, but we should have global value for `ZOOM_F` as we ship apps for the same devices (iphone 6-11, android 7+)

Currently I have very string deadline, but I will work on this functionality later and share my progress.

cristianoccazinsp, 2019-11-03:
Zooming out should start as soon as you change the fingers direction. If
you are seeing a noticeable delay between the user stopping moving his
fingers and the zoom effect change, you are probably using a factor that's
too small and causing too much work to the device. I haven't had issues
from iPhone 7 to 11 with 0.01 as long as the max zoom is set.

Ashoat, 2019-11-03:
@fabriziobertoglio1987 -

The use of `pinchJustEnded` is to make sure that we do a final zoom update once the fingers are released and the pinch gesture concludes. It doesn't detect whether the direction of the pinch changes, but you could definitely improve the Reanimated "code" to do that! If you do, please let me know what you come up with.

When I talk about "bridge chatter", I'm referring to the React Native "bridge" that facilitates communication between the JS and native side. By using `_onPinchProgress` you are asking for a JS update on every "frame" (decided by Gesture Handler), but I avoid that by setting up Reanimated to only update the JS side when the RNCamera zoom factor needs to be updated. This is the optimization I talked about in my previous comment, and it probably only matters for really old Android phones.

Last bit: I'm using a multiplicative factor to decide when the update the zoom factor, ie. when the pinch-gesture zoom is 3% more than the RNCamera zoom, I update. The 3% number is varied [depending on platform](https://github.com/Ashoat/squadcal/blob/master/native/media/camera-modal.react.js#L56-L67). Curious to know whether a multiplicative factor or additive factor makes more sense, and also about how they translate. I would like to optimize my factor by don't have all the devices I'd like to use for research.

fabriziobertoglio1987, 2019-11-22:
Thanks a lot everyone for the advice

I ended up setting the `ZOOM_F` constant to the following levels

```javascript
const ZOOM_F = Platform.OS === 'ios' ? 0.001 : 0.08;
```

The zoom is not fluent as in the default camera app, but it works fine (especially on android)
I will keep improving the functionality and come back here with the sourcecode, but right now I don't have free time.

This is the link to the [ios](https://apps.apple.com/us/app/surfcheck/id1477072496?ls=1) and [android](https://play.google.com/store/apps/details?id=com.surfapp&pcampaignid=MKT-Other-global-all-co-prtnr-py-PartBadge-Mar2515-1) apps (to test the camera functionality in production).

I experienced great performance improvement in the production (google play and apple connect) versions, but the iPhone zoom functionality is still not very fluid.

I agree with Ashoat with the multiplicative factor. 
Maybe not all devices (iPhone and new Samsung Galaxy) use the `zoom` prop the same way?
I used a new Samsung Galaxy and noticed that a single `pitch` will increase the `zoom` from `0x` to `1x` (optical zoom), then the following pitches zoom much less and use the digital zoom... The same applies to the iPhone 6 camera, the first pitch will increase the zoom of a higher proportions then the following ones, as the optical zoom is 1-2 times, then follows the digital zoom.

This would explain the camera behaviour inside my iPhone 6 app, where the first pitch increases the zoom much more then other following pitches, but probably the functionality we wrote does not take that into consideration and the first pitch applies too much zoom (for this reason I had to use `ZOOM_F = 0.001`)

Probably the solution are included in the [ios camera docs](https://developer.apple.com/documentation/avfoundation/cameras_and_media_capture#overview) and I will keep investigating on this topic.

Thanks!

fabriziobertoglio1987, 2020-01-15:
Thanks @Ashoat 

As I am searching for a `first-pull-request` (`Javascript`/`Android`), I am interested in implementing this feature as explained [in my post](https://github.com/react-native-community/react-native-camera/issues/2178#issuecomment-574228101) in the pull request **Pinch to zoom with RN Camera** [#2178](https://github.com/react-native-community/react-native-camera/issues/2178).

@cristianoccazinsp related to [our discussion](https://github.com/react-native-community/react-native-camera/issues/1282#issuecomment-544104090) on `setNativeProps`, as I summarized in this [post](https://github.com/react-native-community/react-native-camera/issues/2178#issuecomment-574228101), I believe `setNativeProps` is available only in the `ReactNative` components (for ex. `View`), but the [`RNCamera`](https://github.com/react-native-community/react-native-camera/blob/master/src/RNCamera.js#L860) component, which is an **interface** between the native java [`CameraViewManager`](https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/CameraViewManager.java#L42) and the react [`Camera`](https://github.com/react-native-community/react-native-camera/blob/master/src/RNCamera.js#L345) component, does not implement `setNativeProps()`.

You can find more details [in my post](https://github.com/react-native-community/react-native-camera/issues/2178#issuecomment-574228101) in the pull request **Pinch to zoom with RN Camera** [#2178](https://github.com/react-native-community/react-native-camera/issues/2178).

I am very happy to receive feedback. Thanks

MostafaMAbdellateif, 2020-01-20:
@fabriziobertoglio1987  did you implement this feature or have any working solution?

SimonErm, 2020-04-30:
#2815

Usamaliaquat123, 2020-11-23:
try  `useNativeZoom={true}`

hungdev, 2021-11-24:
> > try `useNativeZoom={true}`
> 
> Only works on android

you need use this with `maxZoom` props
[image]