react-native-camera/react-native-camera #1284: Could not find com.github.react-native-community:cameraview:d5d9b0d925494ef451ce3eef3fdf14cc874d9baa

eroosenmaallen, 2018-02-23:
FAILURE: Build failed with an exception.

* What went wrong:
Could not resolve all files for configuration ':app:debugCompileClasspath'.
> Could not find com.github.react-native-community:cameraview:d5d9b0d925494ef451ce3eef3fdf14cc874d9baa.
  Searched in the following locations:
      https://jcenter.bintray.com/com/github/react-native-community/cameraview/d5d9b0d925494ef451ce3eef3fdf14cc874d9baa/cameraview-d5d9b0d925494ef451ce3eef3fdf14cc874d9baa.pom
      https://jcenter.bintray.com/com/github/react-native-community/cameraview/d5d9b0d925494ef451ce3eef3fdf14cc874d9baa/cameraview-d5d9b0d925494ef451ce3eef3fdf14cc874d9baa.jar
      https://dl.google.com/dl/android/maven2/com/github/react-native-community/cameraview/d5d9b0d925494ef451ce3eef3fdf14cc874d9baa/cameraview-d5d9b0d925494ef451ce3eef3fdf14cc874d9baa.pom
      https://dl.google.com/dl/android/maven2/com/github/react-native-community/cameraview/d5d9b0d925494ef451ce3eef3fdf14cc874d9baa/cameraview-d5d9b0d925494ef451ce3eef3fdf14cc874d9baa.jar
  Required by:
      project :app > project :../local_modules/react-native-camera/

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

* Get more help at https://help.gradle.org

BUILD FAILED in 6s
25 actionable tasks: 11 executed, 14 up-to-date
Could not install the app on the device, read the error above for details.
Make sure you have an Android emulator running or a device connected and have
set up your Android development environment:
https://facebook.github.io/react-native/docs/android-setup.html

sibelius, 2018-02-23:
Missing jitpack

closed 2018-02-23 by sibelius

eroosenmaallen, 2018-02-23:
From build.gradle:
repositories {
  mavenCentral()
  maven {
   url 'https://maven.google.com'
  }
  maven { url "https://jitpack.io" }
}

Not missing jitpack.