react-native-camera/react-native-camera #1285: Could not find com.github.react-native-community:cameraview:d5d9b0d925494ef451ce3eef3fdf14cc874d9baa

eroosenmaallen, 2018-02-23:
FAILURE: Build failed with an exception.

What went wrong:
Could not resolve all files for configuration ':app:debugCompileClasspath'.
Could not find com.github.react-native-community:cameraview:d5d9b0d925494ef451ce3eef3fdf14cc874d9baa.
Searched in the following locations:
https://jcenter.bintray.com/com/github/react-native-community/cameraview/d5d9b0d925494ef451ce3eef3fdf14cc874d9baa/cameraview-d5d9b0d925494ef451ce3eef3fdf14cc874d9baa.pom
https://jcenter.bintray.com/com/github/react-native-community/cameraview/d5d9b0d925494ef451ce3eef3fdf14cc874d9baa/cameraview-d5d9b0d925494ef451ce3eef3fdf14cc874d9baa.jar
https://dl.google.com/dl/android/maven2/com/github/react-native-community/cameraview/d5d9b0d925494ef451ce3eef3fdf14cc874d9baa/cameraview-d5d9b0d925494ef451ce3eef3fdf14cc874d9baa.pom
https://dl.google.com/dl/android/maven2/com/github/react-native-community/cameraview/d5d9b0d925494ef451ce3eef3fdf14cc874d9baa/cameraview-d5d9b0d925494ef451ce3eef3fdf14cc874d9baa.jar
Required by:
project :app > project :../local_modules/react-native-camera/

Jitpack is included in build.gradle:

```
repositories {
  mavenCentral()
  maven {
   url 'https://maven.google.com'
  }
  maven { url "https://jitpack.io" }
}
```

I'm using a clone of react-native-camera 1.0.1  (under local_modules) with the FaceDetection subdirectory removed to allow it to build for iOS.

sibelius, 2018-02-23:
Try to run RNCamera example, opening another issue won’t solve ur problem

closed 2018-02-23 by sibelius

sibelius, 2018-02-23:
join expo slack if u need more help on react-native-camera