react-native-camera/react-native-camera #1289: Reference example configuration, why error?

yt7649757, 2018-02-26:
` What went wrong:
A problem occurred configuring project ':app'.
> Could not resolve all dependencies for configuration ':app:_debugApkCopy'.
   > Could not find com.github.react-native-community:cameraview:d5d9b0d925494ef451ce3eef3fdf14cc874d9baa.
     Required by:
         cxbd:app:unspecified > cxbd:react-native-camera:unspecified
`

my react-native version: 0.45.1

yt7649757, 2018-02-26:
A problem occurred configuring project ':app'.
> Could not resolve all dependencies for configuration ':app:_debugApkCopy'.
   > Could not find com.android.support:exifinterface:26.0.1.
     Searched in the following locations:
         file:/D:/AndroidSDK/extras/android/m2repository/com/android/support/exifinterface/26.0.1/exifinterface-26.0.1.pom
         file:/D:/AndroidSDK/extras/android/m2repository/com/android/support/exifinterface/26.0.1/exifinterface-26.0.1.jar
         file:/E:/cxbd/android/sdk-manager/com/android/support/exifinterface/26.0.1/exifinterface-26.0.1.jar
     Required by:
         cxbd:app:unspecified

yt7649757, 2018-02-26:
dependencies {
    compile (project(':react-native-camera')) {
            exclude group: "com.android.support"

            // uncomment this if also com.google.android.gms:play-services-vision versions are conflicting
            // this can happen if you use react-native-firebase
            // exclude group: "com.google.android.gms"
        }

        compile ('com.android.support:exifinterface:26.0.1') {
            force = true;
        }
    compile fileTree(dir: "libs", include: ["*.jar"])
    compile "com.android.support:appcompat-v7:23.0.1"
    compile "com.facebook.react:react-native:+"  // From node_modules
}

allprojects {
    repositories {
        maven { url "https://jitpack.io" }
    }
}

<uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.SYSTEM_ALERT_WINDOW"/>
    <uses-permission android:name="android.permission.CAMERA" />
    <uses-permission android:name="android.permission.RECORD_AUDIO"/>
    <uses-permission android:name="android.permission.RECORD_VIDEO"/>
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />

import com.lwansbrough.RCTCamera.RCTCameraPackage;
...
 @Override
    protected List<ReactPackage> getPackages() {
      return Arrays.<ReactPackage>asList(
          new MainReactPackage(),
          new RCTCameraPackage()
      );
    }

rootProject.name = 'cxbd'

include ':react-native-camera'
project(':react-native-camera').projectDir = new File(rootProject.projectDir, 	'../node_modules/react-native-camera/android')

include ':app'

sibelius, 2018-02-26:
Missing exifinterface dep or google maven repository

closed 2018-02-26 by sibelius