react-native-camera/react-native-camera #1292: RNCameraManager crash issues after QRCODE read

jleed, 2018-02-26:
### Steps to reproduce
1. scanning QRCode with weird encoding on something like that
2. - (void)captureOutput:(AVCaptureOutput *)captureOutput didOutputMetadataObjects:(NSArray *)metadataObjects fromConnection:(AVCaptureConnection *)connection  calls
3. metadata.stringValue is NULL that causes application crash.

### Expected behaviour
Tell us what should happen
metadata.stringValue should be not be NULL

### Actual behaviour
Tell us what happens instead
metadata.stringValue is NULL

### Environment
- **Node.js version**: 8.9.1
- **React Native version**: 0.51.0
- **React Native platform + platform version**: iOS 11.2

### react-native-camera
**Version**: npm version 1.0.1

gre, 2018-03-15:
can reproduce with this QRCode:

[image]

(generated on https://smealum.github.io/ninjhax2/ )

gre, 2019-06-01:
@itrcz were you able to still reproduce?

closed 2019-08-07 by stale[bot]