react-native-camera/react-native-camera #1295: Taking a picture with 16:9 ratio causes fatal out of memory crash

Tsiniloiv, 2018-02-27:
### Which implementation are you using

RNCamera

### Steps to reproduce
1. Set camera ratio to "16:9"
2. Take a picture
3. Unfortunately, your app has stopped.

My camera component:
```
<RNCamera
          ref={cam => {
            this.camera = cam;
          }}
          style={styles.preview}
          ratio={this.state.ratio}
          type={this.state.type}
          onCameraReady={this.prepareRatio}
        >
```
My takePicture function:

```
var options = { quality: 1, exif: true};
var directory = RNFS.PicturesDirectoryPath + '/KulaScope/Raw';
const data2 = await this.camera.takePictureAsync();
console.log(data2);
var nameRegExp = /\/(.[^\/]+)$/;
console.log(nameRegExp.exec(data2.uri)[0]);
RNFS.moveFile(data2.uri, (directory + (nameRegExp.exec(data2.uri))[0]).toString()).then(() => console.log('Picture taken and saved'));
} catch (err) {
console.log("takePictureAsync ERROR: " + err);
}
```

### Expected behaviour
A landscape picture is taken.

### Actual behaviour
I'm upgrading from RCTCamera to RNCamera, and now when I try to take a landscape picture my app crashes. A logcat dump turns up an out of memory error. 

```
02-27 13:21:58.577 15662 15678 E AndroidRuntime: java.lang.RuntimeException: An error occurred while executing do
InBackground()
02-27 13:21:58.577 15662 15678 E AndroidRuntime:        at android.os.AsyncTask$3.done(AsyncTask.java:309)
02-27 13:21:58.577 15662 15678 E AndroidRuntime:        at java.util.concurrent.FutureTask.finishCompletion(Futur
eTask.java:354)
02-27 13:21:58.577 15662 15678 E AndroidRuntime:        at java.util.concurrent.FutureTask.setException(FutureTas
k.java:223)
02-27 13:21:58.577 15662 15678 E AndroidRuntime:        at java.util.concurrent.FutureTask.run(FutureTask.java:24
2)
02-27 13:21:58.577 15662 15678 E AndroidRuntime:        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.ja
va:234)
02-27 13:21:58.577 15662 15678 E AndroidRuntime:        at java.util.concurrent.ThreadPoolExecutor.runWorker(Thre
adPoolExecutor.java:1113)
02-27 13:21:58.577 15662 15678 E AndroidRuntime:        at java.util.concurrent.ThreadPoolExecutor$Worker.run(Thr
eadPoolExecutor.java:588)
02-27 13:21:58.577 15662 15678 E AndroidRuntime:        at java.lang.Thread.run(Thread.java:818)
02-27 13:21:58.577 15662 15678 E AndroidRuntime: Caused by: java.lang.OutOfMemoryError: Failed to allocate a 6348
9036 byte allocation with 16769216 free bytes and 44MB until OOM
```

Lowering the quality option avoids this crash. I would prefer not to compromise on quality.

### Environment
- **Node.js version**: 9.3.0
- **React Native version**: 0.47.2
- **React Native platform + platform version**: Android API 23

### react-native-camera
**Version**: npm version

Fconstant, 2018-02-27:
I can't reproduce this issue, using exactly what you stated:
* ratio set to `"16:9"`
* `exif: true`
* `quality: 1`

The only difference is the RN version, mine is `0.53`

Try creating a brand new project and recreate this issue from scratch. Share me with the results.

You can try adding `android:hardwareAccelerated="false"` and `android:largeHeap="true"` to your `application` tag on your Manifest file ([Source](
https://stackoverflow.com/questions/32244851/androidjava-lang-outofmemoryerror-failed-to-allocate-a-23970828-byte-allocatio)). Although I can't assume this is the best solution.

If this stops being a problem, please, don't forget to close the issue.

Tsiniloiv, 2018-02-27:
Problem went away after I assembled a release version. Thanks for the tips, yo.

closed 2018-02-27 by Tsiniloiv