react-native-camera/react-native-camera #1299: RNCamera, Android MediaRecorder stop failed: -2147483648

condorman, 2018-02-28:
### Steps to reproduce
1. react-native init testRnCamera
2. npm install --save git+https://git@github.com/react-native-community/react-native-camera
3. follow configuration for AndroidManifest.xml

### Actual behaviour
`<RNCamera
                            ref={(cam) => this.camera = cam }
                            style = {styles.preview}
                            type={"front"}
                            flashMode={"off"}
                            ratio={"16:9"}
                            permissionDialogTitle={'Test Title'}
                            permissionDialogMessage={'Test Desk'}
/>`

**Test function**
`this.camera.recordAsync()
        .then(data => { console.log(data); })
        .catch(err => console.error(err));  

        setTimeout(()=>{ this.camera.stopRecording(); },4000);

The file is created on device, in path: file:///data/user/0/com.testRnCamera/cache/Camera/886a1949-8ba6-449f-b4e5-2ea055693eea.mp4 with 3,4Mbyte size but:
- when i move on Windows I can't open (with media player or VLC) 
- when i play on device I can't play
- when i try to open with react-native-video i have this error "Unrecognized media format"
I think the movie file is corrupted

**Android log:**
`02-28 11:54:03.154: I/MediaRecorderJNI(7755): stop
02-28 11:54:04.221: E/MediaRecorder(7755): stop failed: -2147483648
02-28 11:54:04.223: W/System.err(7755): java.lang.RuntimeException: stop failed.
02-28 11:54:04.224: W/System.err(7755): 	at android.media.MediaRecorder._stop(Native Method)
02-28 11:54:04.225: W/System.err(7755): 	at android.media.MediaRecorder.stop(MediaRecorder.java:1205)
02-28 11:54:04.225: W/System.err(7755): 	at com.google.android.cameraview.Camera2.stopMediaRecorder(Camera2.java:1058)
02-28 11:54:04.225: W/System.err(7755): 	at com.google.android.cameraview.Camera2.stopRecording(Camera2.java:466)
02-28 11:54:04.225: W/System.err(7755): 	at com.google.android.cameraview.CameraView.stopRecording(CameraView.java:459)
02-28 11:54:04.225: W/System.err(7755): 	at org.reactnative.camera.CameraModule$5.execute(CameraModule.java:252)
02-28 11:54:04.226: W/System.err(7755): 	at com.facebook.react.uimanager.UIViewOperationQueue$UIBlockOperation.execute(UIViewOperationQueue.java:510)
02-28 11:54:04.226: W/System.err(7755): 	at com.facebook.react.uimanager.UIViewOperationQueue$1.run(UIViewOperationQueue.java:819)
02-28 11:54:04.226: W/System.err(7755): 	at com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches(UIViewOperationQueue.java:926)
02-28 11:54:04.226: W/System.err(7755): 	at com.facebook.react.uimanager.UIViewOperationQueue.access$2100(UIViewOperationQueue.java:47)
02-28 11:54:04.226: W/System.err(7755): 	at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:986)
02-28 11:54:04.227: W/System.err(7755): 	at com.facebook.react.uimanager.GuardedFrameCallback.doFrame(GuardedFrameCallback.java:31)
02-28 11:54:04.227: W/System.err(7755): 	at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:136)
02-28 11:54:04.227: W/System.err(7755): 	at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame(ChoreographerCompat.java:107)
02-28 11:54:04.227: W/System.err(7755): 	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:925)
02-28 11:54:04.227: W/System.err(7755): 	at android.view.Choreographer.doCallbacks(Choreographer.java:702)
02-28 11:54:04.228: W/System.err(7755): 	at android.view.Choreographer.doFrame(Choreographer.java:635)
02-28 11:54:04.228: W/System.err(7755): 	at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:913)
02-28 11:54:04.228: W/System.err(7755): 	at android.os.Handler.handleCallback(Handler.java:751)
02-28 11:54:04.228: W/System.err(7755): 	at android.os.Handler.dispatchMessage(Handler.java:95)
02-28 11:54:04.228: W/System.err(7755): 	at android.os.Looper.loop(Looper.java:154)
02-28 11:54:04.229: W/System.err(7755): 	at android.app.ActivityThread.main(ActivityThread.java:6692)
02-28 11:54:04.229: W/System.err(7755): 	at java.lang.reflect.Method.invoke(Native Method)
02-28 11:54:04.229: W/System.err(7755): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1468)
02-28 11:54:04.229: W/System.err(7755): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1358)
`

### Environment
- Nodejs version: 8.9.4
- React Native version: 0.53.3:
- Phisical Device Android 7.0 (Samsung s7+)

### react-native-camera
Master version

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

jgfidelis, 2018-03-02:
probably a bug on the front camera api. Expo is checking on that.

condorman, 2018-03-05:
WIth new master release the problem is solved but the video have vertical flip.

closed 2018-03-07 by condorman