react-native-camera/react-native-camera #1301: [RNCamera] - Focus stops working

stoffern, 2018-02-28:
### Steps to reproduce
1. switch from front to back
2.switch from back to front
3.focus lost..

### Environment
- **Node.js version**: 8.9.4
- **React Native version**: 0.53.3
- **React Native platform + platform version**: Android

### react-native-camera
**Version**: 1.0.1

jgfidelis, 2018-02-28:
Could you test if it happens on the expo version as well?
Use this app: https://github.com/expo/camerja

stoffern, 2018-02-28:
@jgcmarins same problem.
But if i toggle AutoFocus off-on then it will start to focus again.

sibelius, 2018-02-28:
@stoffern could u send a PR to fix this?

stoffern, 2018-02-28:
@sibelius i can always provide a js fix for it.
But Java is not my thing, and it is probably better so solve it natively?

guilhermebruzzi, 2018-03-14:
Same problem with RN 0.43.4 and react-native-camera 1.0.1

closed 2019-06-08 by stale[bot]