react-native-camera/react-native-camera #1302: Barcode scanning does not work on Android 8.0

szlezak, 2018-02-28:
### Steps to reproduce
1. Scanning of barcode 

### Expected behaviour
The camera should recognise the barcode and display scanned information.

### Actual behaviour
The camera does not recognise any code.

### Environment
- **Node.js version**: 9.0.0
- **React Native version**: 0.53.3
- **React Native platform + platform version**: Android 8.0.0

### react-native-camera
**Version**: 1.0.1

sibelius, 2018-02-28:
please provide logs

szlezak, 2018-02-28:
I've just tried to run it trough Android studio and check the logs, they doesn't contain any errors, but the camera is working properly it just cannot recognise the barcode, also I tried it on Android 7 and it works like a charm.

sibelius, 2018-02-28:
try with different aspect ratios

szlezak, 2018-02-28:
Sorry I've missed this prop but thanks for pointing out. It work as intended

closed 2018-02-28 by szlezak

niyas-ali, 2018-04-18:
@szlezak Am also facing the same issue..can u pls tell me what change you have done for fixing this issue??