react-native-camera/react-native-camera #1304: App crashes: RNCameraManager.takePicture was called with 2 arguments but expects 1 arguments.

bvv218, 2018-03-01:
I have updated to RNCamera, and when I take a picture the app crashes with the following error (both android and ios):

```
RNCameraManager.takePicture was called with 2 arguments but expects 1 arguments. If you haven't changed this method yourself, this usually means that your versions of the native code and JavaScript code are out of sync. Updating both should make this error go away.
```

sibelius, 2018-03-01:
Are u using the latest react native version?

bvv218, 2018-03-01:
@sibelius I am using version 0.51.0 of react native

jgfidelis, 2018-03-01:
How is your code?

sibelius, 2018-03-01:
it is getting the wrong react native android version, check your android configs based on https://github.com/react-native-community/rncamera-example

closed 2018-03-01 by sibelius