react-native-camera/react-native-camera #1306: RNCamera  starting Crash

engelsizmuzik, 2018-03-01:
the problem is that when I take more video mode the application never opens and returns an error

import {RNCamera} from 'react-native-camera' I can not use it like this does not debug the application
import RNCamera from 'react-native-camera'   this way I get the following errors and I think I get an error accessing the library

react-native-camera versions 1.0.1
React-native versions 0.53.3

App started  crashed.

this.camera.capture working 
this.camera.capture(mode:video)  crash
takeCapture not working
recordAsync not working crash

sibelius, 2018-03-01:
provide the crash log

engelsizmuzik, 2018-03-02:
03-02 02:28:22.451  2032  2225 W ReactNativeJS: Possible Unhandled Promise Rejection (id: 0):
03-02 02:28:22.451  2032  2225 W ReactNativeJS: Error: recordAsync: Expected a Camera component
03-02 02:28:22.451  2032  2225 W ReactNativeJS: createErrorFromErrorData@http://10.0.3.2:8081/index.delta?platform=android&dev=true&minify=false:2016:26
03-02 02:28:22.451  2032  2225 W ReactNativeJS: http://10.0.3.2:8081/index.delta?platform=android&dev=true&minify=false:1968:51
03-02 02:28:22.451  2032  2225 W ReactNativeJS: __invokeCallback@http://10.0.3.2:8081/index.delta?platform=android&dev=true&minify=false:2410:23
03-02 02:28:22.451  2032  2225 W ReactNativeJS: http://10.0.3.2:8081/index.delta?platform=android&dev=true&minify=false:2155:34
03-02 02:28:22.451  2032  2225 W ReactNativeJS: __guardSafe@http://10.0.3.2:8081/index.delta?platform=android&dev=true&minify=false:2323:13
03-02 02:28:22.451  2032  2225 W ReactNativeJS: invokeCallbackAndReturnFlushedQueue@http://10.0.3.2:8081/index.delta?platform=android&dev=true&minify=false:2154:21
03-02 02:28:22.451  2032  2225 W ReactNativeJS: invokeCallbackAndReturnFlushedQueue@[native code]

nbennink, 2018-03-12:
I have the "recordAsync: Expected a Camera component" as well.

Is there already a fix for this? I'm using RNCamera 1.02 with React Native 0.53. The issue is Android only for me.

internet5, 2018-03-19:
I have this problem too

jdeanwaite, 2018-03-23:
Also receiving this error on Android while trying to record a video.
```
Error: recordAsync: Expected a Camera component
Error: recordAsync: Expected a Camera component
    at createErrorFromErrorData (/Users/justin/VSCodeProjects/MyApp/.vscode/.react/index.bundle:2026:17)
    at /Users/justin/VSCodeProjects/MyApp/.vscode/.react/index.bundle:1978:27
    at MessageQueue.__invokeCallback (/Users/justin/VSCodeProjects/MyApp/.vscode/.react/index.bundle:2420:18)
    at /Users/justin/VSCodeProjects/MyApp/.vscode/.react/index.bundle:2165:18
    at MessageQueue.__guardSafe (/Users/justin/VSCodeProjects/MyApp/.vscode/.react/index.bundle:2333:11)
    at MessageQueue.invokeCallbackAndReturnFlushedQueue (/Users/justin/VSCodeProjects/MyApp/.vscode/.react/index.bundle:2164:14)
    at /Users/justin/VSCodeProjects/MyApp/.vscode/.react/debuggerWorker.js:126:58
    at process.<anonymous> (/Users/justin/VSCodeProjects/MyApp/.vscode/.react/debuggerWorker.js:35:9)
    at emitTwo (events.js:126:13)

```

jdeanwaite, 2018-03-23:
This turned out to be a permissioning issue. I made sure to update `AndroidManifest.xml` to include the  following permissions: 
```
  <uses-permission android:name="android.permission.CAMERA" />
  <uses-permission android:name="android.permission.RECORD_AUDIO"/>
  <uses-permission android:name="android.permission.RECORD_VIDEO"/>
  <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
  <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
```

@engelsizmuzik Have you added these permissions as well?

Sangrene, 2018-03-24:
Got this issue as well, with exactly the same environment on an Android device Moto G4.

Error is the following :
```
03-24 16:38:18.363 11341 11404 I ReactNativeJS: Starting record
03-24 16:38:18.429 11341 11404 I ReactNativeJS: 'Error recording :', { [Error: recordAsync: Expected a Camera component]
03-24 16:38:18.429 11341 11404 I ReactNativeJS:   framesToPop: 1,
03-24 16:38:18.429 11341 11404 I ReactNativeJS:   code: 'E_CAMERA_BAD_VIEWTAG',
03-24 16:38:18.429 11341 11404 I ReactNativeJS:   line: 14461,
03-24 16:38:18.429 11341 11404 I ReactNativeJS:   column: 31,
03-24 16:38:18.429 11341 11404 I ReactNativeJS:   sourceURL: 'http://localhost:8081/index.delta?platform=android&dev=true&minify=false' }
```

Sangrene, 2018-03-25:
Ok I fixed this this way : 
It will ask the audio record permission when you try to record a video without muting it for the first time. The record won't start but it will throw an exception and you can use it to handle things your way. This is just a quick workaround, I hope this will be fixed ASAP :)

Add this code to the beginning of the **record** method in the **CameraModule** class :

```
if(Build.VERSION.SDK_INT >= Build.VERSION_CODES.M && options.hasKey("mute") && ContextCompat.checkSelfPermission(getCurrentActivity(), Manifest.permission.RECORD_AUDIO) != PackageManager.PERMISSION_GRANTED){

          getCurrentActivity().requestPermissions(new String[]{Manifest.permission.RECORD_AUDIO}, 123);
          promise.reject("E_CAMERA_RECORD_AUDIO_NOT_GRANTED", "Record audio permission has been asked but not granted yet");
          return;
      }

```

SimonErm, 2018-09-25:
RCTCamera is deprecated, use RNCamera instead. Please open a new issue if this problem still occures.

closed 2018-09-25 by SimonErm

rajvir13, 2020-08-23:
I am getting the issue while passing path in option in recordAsync method.
Any solution to this as I am not able to video on returned uri by the method. that's why I was trying to pass path as option in method.

Thanks in advance