react-native-camera/react-native-camera #1307: onBarCodeRead event no longer has bounds

chris-247-sloan, 2018-03-01:
**Which implementation are you using**

I've just converted from RCTCamera to RNCamera

**Steps to reproduce**

i followed the advised steps [here](https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md) 

**Expected behaviour**
Previously, in RCTCamera, the onBarCodeRead event had a bounds property. This allowed my app to use the bounds to highlight on the screen when it had detected a barcode. (QR Codes specifically) But now this no longer exists. It's no longer documented either. (and so I guess its a change that was planned rather than a bug) 

What I'd like to know is: 

- What was the reasoning for this no longer being part of the event object? 
- Was I doing something wrong by using this?
- Are there plans to add it back into a future release? 
- Is there some other possibility to enable similar functionality in my app? 

**Actual behaviour**
Red error screen with "Cannot read property 'origin' of undefined.

**Environment**
_taken from package.json_ 
    "react-native": "0.52.0",
    "react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera",.

Thanks

chris-247-sloan, 2018-03-08:
Is there anyone able to answer my questions?

Fconstant, 2018-03-28:
Done by #1402

closed 2018-03-28 by Fconstant