react-native-camera/react-native-camera #1310: [RNCamera] - Android crash on init (FaceDetector$Builder)

jpudysz, 2018-03-02:
### Which implementation are you using

*RNCamera*

### Steps to reproduce
1. Use simulator or phone with Android
2. Use RNCamera
3. Navigate to camera page

### Expected behaviour
Should work as on iOS or like deprecated RCTCamera

### Actual behaviour
Exception

### Environment
- **Node.js 8.5**:
- **React Native 0.53.3**:
- **Android 7.0**: 

### react-native-camera
**Version**: 1.0.1

[image]

sibelius, 2018-03-02:
U need to include google Visio dep on ur gradle config

closed 2018-03-02 by sibelius

jpudysz, 2018-03-02:
Ah thanks! I missed it 👍 

For future reference, take a look at:
```
compile ("com.google.android.gms:play-services-vision:10.2.0") {
  force = true;
}
```

In my case I had a lot of other conditions like `play-services-base` , `play-services-maps` etc. and I missed that in docs we have `play-services-vision` :)