react-native-camera/react-native-camera #1311: takePictureAsync never returns when children elements are updated

ospfranco, 2018-03-02:
*RNCamera*

### Steps to reproduce
1. <Camera> Component with children element inside
2. on takePictureAsync use setState to change one of the children element (ex. conditional rendering)
3. takePictureAsync will never return

### Expected behaviour
Picture should be taken normally and UI updated correctly

### Actual behaviour
takePictureAsync never returns

### Environment
- 0.53
- Android (tested on android 7)

### react-native-camera
**1.0.1**

sibelius, 2018-03-02:
does this happens on expo camera module as well?

ospfranco, 2018-03-28:
hey sorry, missed the message, not really sure, didn't have time to test this on expo

pstanton, 2018-05-31:
#1373

hayo0914, 2018-06-25:
I'm using expo, and having same trouble.
Expo=v27.0.1
ReactNative=0.55.2

IssuehuntBot, 2018-11-19:
@0maxxam0 funded this issue with $20. [See it on IssueHunt](https://issuehunt.io/repos/33218414/issues/1311)

closed 2019-01-08 by sibelius

IssuehuntBot, 2019-01-08:
@thymikee has submitted a pull request. [See it on IssueHunt](https://issuehunt.io/repos/33218414/issues/1311)

n1ru4l, 2019-01-08:
:tada: This issue has been resolved in version 1.7.0 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v1.7.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

IssuehuntBot, 2019-01-08:
@sibelius has rewarded $14.00 to @thymikee. [See it on IssueHunt](https://issuehunt.io/repos/33218414/issues/1311)

- :moneybag: Total deposit: $20.00
- :tada: Repository reward(20%): $4.00
- :wrench: Service fee(10%): $2.00