react-native-camera/react-native-camera #1313: Barcode scanning not working except QR CODE ( Android )

gimwee, 2018-03-03:
Try to scan EAN_13 Barcode , works on ios but not android

Tested with Note 8: Android 7.1.1

react-native-camera
Version: 1.0.1

huangguozhen, 2018-03-03:
@gimwee must use `import Camera from 'react-native-camera'`

gimwee, 2018-03-03:
is the usage same as RNCamera?

gimwee, 2018-03-03:
thanks @huangguozhen , your method works

closed 2018-03-03 by gimwee

abdulrahman-khankan, 2018-03-03:
@huangguozhen thanks for your comment!

Example of simple Alert popup when a barcode is detected.

Imports
``` App.js
import Camera from 'react-native-camera'
import { Alert } from 'react-native';
```
Barcode handler
``` App.js
  BarCodeRead({data}) {
    console.log(data)
    Alert.alert(data)
  }
```
render
``` App.js
<View style={styles.container}>
  <Camera
    ref={ref => { this.camera = ref; }}
    style = {styles.preview}
    permissionDialogTitle={'Permission to use camera'}
    permissionDialogMessage={'We need your permission to use your camera phone'}
    onBarCodeRead={this.BarCodeRead.bind(this)}
  />
</View>
```
Styles
``` App.js
const styles = StyleSheet.create({
  container: {
    flex: 1,
    flexDirection: 'column',
    backgroundColor: 'black'
  },
  preview: {
    flex: 1,
    justifyContent: 'flex-end',
    alignItems: 'center'
  },
});
```