react-native-camera/react-native-camera #1314: Is it possible to detect faces while recording video?

rexlow, 2018-03-04:
It appears that face detection has stopped running as soon as `recordAsync()` started. Is there a way to detect faces while recording?
```
recordVideo = async = () => {
    if (this.camera) {
      this.camera.recordAsync({
        quality: "RNCamera.Constants.VideoQuality.720p",
        mute: true
      })
      .then(data => this.uploadMedia(data.uri))
      .catch(error => console.log(error))
    }
  }

stopRecord = async = () => {
  if (this.camera) {
      this.camera.stopRecording()
  }
}

renderCamera() {
    return (
      <RNCamera
        ref={ref => {
          this.camera = ref;
        }}
        style={{
          flex: 1,
        }}
        type={this.state.type}
        flashMode={this.state.flash}
        autoFocus={this.state.autoFocus}
        ratio={this.state.ratio}
        captureQuality={"high"}
        onCameraReady={this.onCameraReady}
        faceDetectionLandmarks={RNCamera.Constants.FaceDetection.Landmarks.all}
        onFacesDetected={this.onFacesDetected}
        onFaceDetectionError={this.onFaceDetectionError}
        permissionDialogTitle={'Permission to use camera'}
        permissionDialogMessage={'We need your permission to use your camera phone'}
      >
        <TouchableOpacity
          style={[styles.flipButton]}
          onPress={this.state.recording === false ? this.recordVideo : this.stopRecord}>
          <Text style={styles.flipText}>{this.state.recording === false ? "START" : "STOP"}</Text>
        </TouchableOpacity>
        {this.renderFaces()}
        {this.renderLandmarks()}
      </RNCamera>
    );
  }
```

jgfidelis, 2018-03-05:
Currently no, when a video starts the facedection module is deactivated.

sibelius, 2018-03-05:
we gonna accept a PR to solve this

io-pan, 2019-02-25:
Hi there, 
Any news on that issue ?
Face detection not only stops running as soon as `recordAsync()` start.
It also doesn't restart  when stopRecording() has been executed.

Suhas404, 2019-12-19:
@rexlow Did you find any solution? Please let us know!

lucastonon, 2020-05-23:
> Currently no, when a video starts the facedection module is deactivated.

Is this a technical limitation, or can I do it myself?

closed 2020-07-30 by stale[bot]

liqteqmehdi, 2021-07-13:
> Currently no, when a video starts the facedection module is deactivated.

any updates for it, can we do it now?