react-native-camera/react-native-camera #1324: Replacement for Capture Target

arapocket, 2018-03-07:
### Steps to reproduce
1. Try to follow spotty documentation and successfully install RNCamera after using RCTCamera
2. Rely on captureTarget={Camera.constants.CaptureTarget.disk} prop to upload files somewhere
3. Get stuck after hours of fixing the million errors I was getting after updating

So if we need to upload the file somewhere what do we do now without the CaptureTarget prop?

sibelius, 2018-03-07:
Upload where?

arapocket, 2018-03-07:
What I am asking is if it's still possible to retrieve the file without using CaptureTarget.disk, because I remember trying to get the file from the cache was an issue when I had tried it before.

arapocket, 2018-03-07:
Nevermind it still works fine.

closed 2018-03-07 by arapocket