react-native-camera/react-native-camera #1325: Android front camera vertical Flip problem

condorman, 2018-03-07:
**Steps to reproduce**

- react-native init testRnCamera
- npm install --save git+https://git@github.com/react-native-community/react-native-camera
- follow configuration for AndroidManifest.xml

**Actual behaviour**
```
<RNCamera ref={(cam) => this.camera = cam }
                     style = {styles.preview} 
                     type={RNCamera.Constants.Type.front}
                     flashMode={RNCamera.Constants.FlashMode.off}
                     ratio={"16:9"} />

```
**Test function** 
```
this.camera.recordAsync()
.then(data => { console.log(data); })
.catch(err => console.error(err));
setTimeout(()=>{ this.camera.stopRecording(); },4000);
```

**Problem**
Video camera preview works correctly, functions and writing of video file on device is working properly but video is vertically inverted.

Environment
Nodejs version: 8.9.4
React Native version: 0.54.0:
Phisical Device: Android 7.0 (Samsung s7+)
react-native-camera: Master version

sibelius, 2018-03-07:
This is the default behavior, check native camera app

condorman, 2018-03-07:
thank you, the default behavior of video recorded is a VERTICAL mirror? it seems strange.. the video produced is upside down. I cant' find information for switch mirror on doc: https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md

thank you, i try to debug JAVA library

sibelius, 2018-03-07:
maybe it is a bug instead, misread your issue

condorman, 2018-03-14:
I find same problem on android cameraview.
[https://github.com/google/cameraview/issues/84](url)

And same problem on Expo:
[https://github.com/expo/expo/issues/1344](url)

At this point I hope this problem is solved on external library.

sibelius, 2018-03-14:
I think u can try this prop `forceUpOrientation`

condorman, 2018-03-14:
Thank you, i try but not work, i think is a prop for IOS in TakePictureOptions. My problem is the saved video on device is Upside-down. i think is a problem of android.cameraview library :\

condorman, 2018-03-21:
I @sibelius, i find the problem in cameraview repository, with [this change ](https://github.com/react-native-community/cameraview/pull/3) the recorded video with FRONT camera in PORTRAIT mode work fine! my issues is solved, thank you!

rkostrab, 2018-12-13:
This PR still not resolved the issue. 
Using `Google Pixel` - Every record from front camera is flipped upside down.
My dependency is `Master`: 
```
"react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera.git"
```

rkostrab, 2018-12-13:
If I put this property co Camera:
```js
orientation={Camera.constants.Orientation.portraitUpsideDown}
```
then the video is OK. But it's bad to set it "portraitUpsideDown" to see it properly.

n1ru4l, 2018-12-13:
@rkostrab Could you please test if https://github.com/react-native-community/react-native-camera/pull/1992 solves this issue?

closed 2018-12-13 by n1ru4l

rkostrab, 2018-12-13:
@n1ru4l,

tested your fix using this dependency (I hope it's correct):
```
"react-native-camera": "git+https://github.com/n1ru4l/react-native-camera.git#feature-android-use-device-orientation"
```

**And results are:**

RCTCamera:
- nothing happend, mp4 still flipped upside down (in: front camera video recording)

RNCamera:
- OK

rkostrab, 2018-12-13:
@n1ru4l No, but I've made some native changes in my fork and now I have to dig for a while in that again and also rewrite JSX... Maybe new `RNCamera` props will enough for my needs.
Maybe it's time to switch to new RNCamera and forget old RCTCamera.

Anyway thanks!