react-native-camera/react-native-camera #1327: RNCamera doesn't support cropToPreview

sunweiyang, 2018-03-07:
The legacy RCTCamera implementation supported cropToPreview as a prop. Can RNCamera get this functionality as well?

jgfidelis, 2018-03-08:
Would be a nice feature to have if you implement it, send a PR

404sand808s, 2018-05-01:
This feature would be very useful (and is a basic must-have IMO). I've had users complain that they image captured is not the same as the viewfinder showed.

jkdf2, 2018-05-15:
References #1057 where `cropToPreview` was implemented for `RCTCamera`

n1ru4l, 2018-09-24:
Ongoing Pull Request: https://github.com/react-native-community/react-native-camera/pull/1636

sibelius, 2019-04-18:
RCTCamera is removed https://github.com/react-native-community/react-native-camera/issues/1890

Migrate to latest version of RNCamera

jkdf2, 2019-04-18:
@sibelius, I am not sure this issue was resolved. This issue is in relation to RNCamera, not the deprecated RCTCamera.

gdoudeng, 2019-10-13:
we need `cropToPreview`

blakito, 2020-04-02:
> we need `cropToPreview`

We do need it too

Luminous13, 2020-06-02:
Any update on this one?

closed 2020-08-08 by stale[bot]

D4nt4lion, 2020-11-18:
Feature is truly needed. Is this library dead?

dipeshkoirala21, 2021-03-01:
Has anyone solved this issue?