react-native-camera/react-native-camera #1330: [Feature Request] iOS Allow setting the codec

n1ru4l, 2018-03-08:
### Which implementation are you using

*RNCamera*

### Description

iOS 11 introduces a new codec (HEVC) which is the default one. This one is incompatible with prior iOS versions. If you have to support multiple devices and do not want to convert the video on your server, it would be very beneficial if we had an option to specify the output format. (Similar to how the capture quality is determined).

Since iOS 10 it is possible to specify the output format: https://developer.apple.com/documentation/avfoundation/avcapturemoviefileoutput/1388448-setoutputsettings?language=objc

If you agree I would love to file in a pull request for this. In the meantime I am going to use my own fork of this library 😅.

sibelius, 2018-03-08:
Please send a PR

n1ru4l, 2018-03-08:
I will send one this weekend.

In the meantime people can workaround this "issue" by using `patch-package` to enforce a specific codec:

`patches/react-native-camera+1.0.1.patch`
```diff
patch-package
--- a/node_modules/react-native-camera/ios/RN/RNCamera.m
+++ b/node_modules/react-native-camera/ios/RN/RNCamera.m
@@ -395,7 +395,9 @@ - (void)record:(NSDictionary *)options resolve:(RCTPromiseResolveBlock)resolve r
         
         AVCaptureConnection *connection = [self.movieFileOutput connectionWithMediaType:AVMediaTypeVideo];
         [connection setVideoOrientation:[RNCameraUtils videoOrientationForInterfaceOrientation:[[UIApplication sharedApplication] statusBarOrientation]]];
-        
+        if (@available(iOS 10, *)) {
+            [self.movieFileOutput setOutputSettings:@{AVVideoCodecKey:AVVideoCodecH264} forConnection:connection];
+        }
         dispatch_async(self.sessionQueue, ^{
             NSString *path = [RNFileSystem generatePathInDirectory:[[RNFileSystem cacheDirectoryPath] stringByAppendingString:@"Camera"] withExtension:@".mov"];
             NSURL *outputURL = [[NSURL alloc] initFileURLWithPath:path];
```

jgfidelis, 2018-03-08:
That would be good to have. If I can suggest, export the constants for the available codecs and create a prop called `ouputFormat` and then you could use `outputFormat=RNCamera.Constants.OutputCodecs.H264}`

Thanks!

n1ru4l, 2018-03-17:
This was implemented in #1339

closed 2018-03-17 by n1ru4l