react-native-camera/react-native-camera #1332: takePicture method is not invoked somehow

suhailjain, 2018-03-08:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Which implementation are you using

*RNCamera* or RCTCamera (RCTCamera will be removed on v2.0.0)?

### Steps to reproduce
code i used : 

  takePicture = async function() {
      if (this.camera) {
        const options = { quality: 0.5, base64: true };
        const data = await this.camera.takePictureAsync(options)
       console.log(data.uri);
    }
   };

  render() {
    return (
      <View style={styles.container}>
        <RNCamera
          ref={(cam) => {
            this.camera = cam;
          }}
          onBarCodeRead={this.onBarCodeRead.bind(this)}
          style={styles.preview}
          type={this.props.type}
          captureAudio={false}
        >
        <View style={styles.innerContainer}>

        <Icon
        raised
        iconStyle={{ backgroundColor: 'transparent' }}
        name='chevron-left'
        color='#003366'
        onPress={() => Actions.pop()}
        />

        <Icon
        raised
        name='camera'
        color='#003366'
        onPress={() => this.takePicture}
        />
        <Icon
        raised
        name='close'
        color='#003366'
        onPress={() => Actions.pop()}
        />

        </View>
        </RNCamera>
      </View>
  

### Expected behaviour
log statement should print to console 

### Actual behaviour
log is never printed thus takePicture is never invoked.
instead i get this in my dev tools console on chrome : 

[RNCamera initializeCaptureSessionInput]_block_invoke: Error Domain=AVFoundationErrorDomain Code=-11814 "Cannot Record" UserInfo={NSLocalizedDescription=Cannot Record, NSLocalizedRecoverySuggestion=Try recording again.}

### Environment
- **Node.js version**:8.9.3
- **React Native version**:0.51
- **React Native platform + platform version**: iOS 11.2
### react-native-camera
**Version**: v1.0.0

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

jgfidelis, 2018-03-08:
https://stackoverflow.com/questions/24176738/error-domain-avfoundationerrordomain-code-11814-cannot-record

Use a device

suhailjain, 2018-03-09:
not working on a real android device.

suhailjain, 2018-03-09:
**Attempted to transition from state `RESPONDER_INACTIVE_PRESS_IN` to `RESPONDER_ACTIVE_LONG_PRESS_IN`, which is not supported. This is most likely due to `Touchable.longPressDelayTimeout` not being cancelled**

above is printed to console when running on android real device and performing repetitive calls to takePicture method.

Please help.

jgfidelis, 2018-03-09:
This is probably a problem not related to RNCamera, I cannot help you.

suhailjain, 2018-03-10:
aint there issues on RNCamera not working for real devices ?

suhailjain, 2018-03-10:
working fine now, JS code was the culprit

closed 2018-03-10 by suhailjain

dann1609, 2018-03-19:
@suhailjain what did you did? I have this problem but only on android, in iOs it works great!.

suhailjain, 2018-03-24:
i changed the way i am declaring and calling the method 'takePicture'.

code changed that worked for me : 

instead of:
takePicture = async function() 

do this:
  async takePicture() {

and instead of:
 onPress={() => this.takePicture

do this:
onPress={() => this.takePicture()

@dann1609  do tell me if it works.

dann1609, 2018-03-24:
@suhailjain my problem was I set state while app is takin picture. so I changed my code to call a set state with the promised returned from camera. No it works great. Also I found I have some premises issues that explain why it work in some androids and not in another.

gshoanganh, 2019-12-31:
Seem you using an emulator, i think you need use 

> an device

.
Hope it helps you