react-native-camera/react-native-camera #1334: onBarCodeRead works excellently on qrcodes but not not on code-128 or code-39

SaiKDot, 2018-03-09:
```
         <RNCamera
            ref={ref => {
              this.camera = ref;
            }}
            style = {styles.preview}
            type={RNCamera.Constants.Type.back}
            flashMode={RNCamera.Constants.FlashMode.on}             
            permissionDialogTitle={'Permission to use camera'}
            permissionDialogMessage={'We need your permission to use your camera phone'}
             barCodeTypes={[RNCamera.Constants.BarCodeType.code39]}
             onBarCodeRead={this._bacr.bind(this)}

        />
```
Nothing really happens. But when I remove the barCodeTypes props it scans qr codes,but not barcodes.
### Environment
- **Node.js version**: 5.5.1
- **React Native version**:  0.54.0
 

### react-native-camera
**Version**: npm version

movie4, 2018-03-09:
ean13 barcode also not working

movie4, 2018-03-09:
Seems like this issue https://github.com/react-native-community/react-native-camera/issues/1334

SaiKDot, 2018-03-09:
I have solved this issue by using <Camera /> instead of <RNCamera />
Anyone reading this, don't forget to add 
`import Camera from 'react-native-camera';`  instead of `import { RNCamera } from 'react-native-camera';`

closed 2018-03-09 by SaiKDot